whoenig/crazyflie_ros #2: Setting up the communication with crazyflie.

batmanatucsd, 2015-04-16:
Hi. We are trying to use the ros nodes to set up the server and connect to a crazyflie, however we can't get a connection.
We are running:
roslaunch crazyflie crazyflie_server.launch
roslaunch crazyflie crazyflie_add.launch uri:=radio://0/10/250K

But the callback to connected, is not called and ROS never prints "connected" to the screen. 
So we are not able to get any /imu msgs from the crazyflie. 
We are however able to publish cmd_vel msgs and make the rotors turn. 

Can you help us with this? We spent many hours looking at the ros code and the crazyflie python apis, but couldn't really figure it out. 
Thank you in advance.

whoenig, 2015-04-16:
Hi, I am happy to help out, but would need some more information:
1.) What crazyflie-clients-python version are you using (branch and if 
possible commit id)
2.) Which ROS version are you using (I assume Indigo?)
3.) Are you working with Crazyflie 1.0 or Crazyflie 2.0?
4.) Does the basiclog.py example work for you?

If the rotors turn, it means that the connection succeeded. As you 
described, it looks like the callback doesn't get triggered. I saw that 
only once and that was a bug in the SDK (which was fixed a few months ago).

batmanatucsd, 2015-04-21:
Hi thank you for replying so fast, and sorry for replying back so late. 
1.
This is the crazy-clients-python that we are using (i guess the latest commit version on github)
https://github.com/bitcraze/crazyflie-clients-python

2. We are using Indigo for ROS

3. Our Crazyflie is Nano, which I think is 1.0

4. I don't know what basiclog.py is.. Can you elaborate on that?

Thank you so much in advance.

batmanatucsd, 2015-04-22:
Okay. I found the basiclog.py and tried running it, but it stops at the same spot as the when running the ROS server node. The green light turns on on the crazyflie, but the callback is never called.

whoenig, 2015-04-22:
Can you make sure that you have the latest crazyflie-clients-python? I.e. if you are using git, use "git pull" to get the latest version. There used to be a bug which was fixed January 19 this year, which prevented logging on the Crazyflie 1.0. I haven't tested with the Crazyflie 1.0 in a while, but I can check on my side as well.

batmanatucsd, 2015-04-22:
I got it to work, but I didn't used the API on the github. I used the host software from https://wiki.bitcraze.io/projects:crazyflie:binaries:index. I think it's an older version but I was able to get the callback after changing some code both in the ROS code and the API. I still don't know why it didn't work in the first place. If you ever figure it, please let me know.

whoenig, 2015-07-28:
The latest version does not use the SDK anymore, so I'll close this issue.

closed 2015-07-28 by whoenig