whoenig/crazyflie_ros #3: multi_teleop connection issues

mechAndy, 2015-05-06:
I am having troubles connecting to multiple quadrotors using your ROS node. I have been following your comments on the bitcraze blog:

https://www.bitcraze.io/2015/04/towards-crazyswarms-2/

I have downloaded your radio_multi-link python client and my radio and crazyflies have the most up-to-date firmware. I am currently running on ROS hyrdo, but will shortly be upgrading to Indigo, on Ubuntu 12.04.

When I run the mutli_teleop_xbox360 node I get the following errors over and over:

[INFO] [WallTime: 1430944581.092314] Connecting to radio://0/80/2M
[INFO] [WallTime: 1430944581.092744] Connecting to radio://0/72/2M
[FATAL] [WallTime: 1430944581.093472] Connection to radio://0/72/2M failed: Couldn't load link driver: Link already open!

Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/cflib/crazyflie/__init__.py", line 217, in open_link
    self._link_error_cb)
  File "/usr/local/lib/python2.7/dist-packages/cflib/crtp/__init__.py", line 90, in get_link_driver
    instance.connect(uri, link_quality_callback, link_error_callback)
  File "/usr/local/lib/python2.7/dist-packages/cflib/crtp/radiodriver.py", line 110, in connect
    raise Exception("Link already open!")
Exception: Link already open!

Thanks!

whoenig, 2015-05-06:
Hi Andy,

It looks like the ROS package is not picking up my radio_multilink 
branch but rather some other installed version (without multi-link) 
support. What do you specify as crazyflieSDK? (Either change it in the 
launch file or use "roslaunch crazyflie_demo multi_teleop_xbox360.launch 
crazyflieSDK:=<path/to/local/version/of/https://github.com/whoenig/crazyflie-clients-python/tree/radio_multilink>"

I hope that helps!

Wolfgang

mechAndy, 2015-05-06:
Hi Wolfgang,

Thanks for the response. Apparently when I pulled your crazyflie python client I forgot to pull from the correct branch.

Thanks for your quick response!

Andy

closed 2015-05-06 by mechAndy