whoenig/crazyflie_ros #4: multirobot flight

mechAndy, 2015-05-29:
How do I use multiple crazy radios for multiple sets of quads?

whoenig, 2015-06-01:
Short Version: Just use a different radio index in the URI.

Longer:
1.) Make sure that you configure you CFs with different addresses, as 
described here: https://www.bitcraze.io/2015/04/towards-crazyswarms-2/.
2.) In your launch file, for each group of quads, use a different radio 
index, e.g. "radio://0/80/250K/0xE7E7E7E701", 
"radio://0/80/250K/0xE7E7E7E702", "radio://1/90/250K/0xE7E7E7E703", 
"radio://1/90/250K/0xE7E7E7E704". There are two quads per radio. The 
quads which share a radio would ideally share channel and datarate as 
well. Different radios should have separate channels, otherwise you will 
get congestions. Depending on your setpoint update rate, you can have up 
to 4 CFs per radio.
3.) Make sure that you either use my custom python-SDK 
(https://github.com/whoenig/crazyflie-clients-python/tree/radio_multilink) 
or use the cpp-branch of crazyflie_ros 
(https://github.com/whoenig/crazyflie_ros/tree/cpp). Please note that 
the cpp-version has less features at the moment (no support for 
logging/parameters), but is self-contained and achieves better tracking 
performance when using several CFs.

mechAndy, 2015-06-03:
I am using your ROS code and python tree without issue. However, I am not able to change the radio address from the 2.0 config panel as shown in the image.

[image]

I also tried to just change the radio ID to match your note, with a single radio, as:
default="radio://0/80/2M/0xE7E7E7E7E7"
and I get the following ROS error:
[INFO] [WallTime: 1433366232.575451] Connecting to radio://0/80/2M/0xE7E7E7E7E7
[FATAL] [WallTime: 1433366232.577298] Connection to radio://0/80/2M/0xE7E7E7E7E7 failed: No driver found or malformed URI: radio://0/80/2M/0xE7E7E7E7E7

Am I missing something stupid?

whoenig, 2015-06-15:
The first issue (not able to change the address) sounds like you don't have the latest firmware on the Crazyflie. Did you upgrade with the latest version on github?

The second issue sounds like you have an old version (or several installed versions) of the Bitcraze SDK. The ROS driver just appends the specified folder the the python search path, so if there is an old version in the search path it finds first, it wont work.

whoenig, 2015-07-28:
The latest version does not use the SDK anymore, so I'll close this issue.

closed 2015-07-28 by whoenig

mechAndy, 2015-08-04:
Sorry for my delayed response. I have stepped away from that project for a little while.

Thanks for your help.