whoenig/crazyflie_ros #5: ImportError: No module named cflib ??

rs1990, 2015-07-02:
Hello Whenever i try to run the launch file i get the following error. Any advice ???

Thank you.

maverick@maverick:~/indigo_ws$ roslaunch crazyflie_demo teleop_xbox360.launch
WARNING: Package "cfclient" does not follow the version conventions. It should not contain leading zeros (unless the number is 0).
... logging to /home/maverick/.ros/log/a31abbc2-20df-11e5-977f-d0df9a8cc6bf/roslaunch-maverick-17184.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

WARNING: Package "cfclient" does not follow the version conventions. It should not contain leading zeros (unless the number is 0).
WARNING: Package "cfclient" does not follow the version conventions. It should not contain leading zeros (unless the number is 0).
WARNING: Package "cfclient" does not follow the version conventions. It should not contain leading zeros (unless the number is 0).
WARNING: Package "cfclient" does not follow the version conventions. It should not contain leading zeros (unless the number is 0).
started roslaunch server http://localhost:52025/

SUMMARY
========

PARAMETERS
 * /crazyflie1/crazyflie_add/enable_logging: True
 * /crazyflie1/crazyflie_add/pitch_trim: 0
 * /crazyflie1/crazyflie_add/roll_trim: 0
 * /crazyflie1/crazyflie_add/tf_prefix: crazyflie1
 * /crazyflie1/crazyflie_add/uri: radio://0/80/2M
 * /crazyflie1/joy/dev: /dev/input/js0
 * /crazyflie1/quadrotor_teleop/x_axis: 5
 * /crazyflie1/quadrotor_teleop/x_velocity_max: 30
 * /crazyflie1/quadrotor_teleop/y_axis: 4
 * /crazyflie1/quadrotor_teleop/y_velocity_max: -30
 * /crazyflie1/quadrotor_teleop/yaw_axis: 1
 * /crazyflie1/quadrotor_teleop/yaw_velocity_max: -200
 * /crazyflie1/quadrotor_teleop/z_axis: 2
 * /crazyflie1/quadrotor_teleop/z_velocity_max: 60000
 * /crazyflie_server/crazyflieSDK: ~/crazyflie/crazy...
 * /rosdistro: indigo
 * /rosversion: 1.11.13

NODES
  /crazyflie1/
    crazyflie_add (crazyflie/crazyflie_add.py)
    crazyflie_demo_controller (crazyflie_demo/controller.py)
    joy (joy/joy_node)
    quadrotor_teleop (hector_quadrotor_teleop/quadrotor_teleop)
  /
    crazyflie_server (crazyflie/crazyflie_server.py)
    rqt_plot (rqt_plot/rqt_plot)
    rqt_plot2 (rqt_plot/rqt_plot)
    rviz (rviz/rviz)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
WARNING: Package "cfclient" does not follow the version conventions. It should not contain leading zeros (unless the number is 0).
process[crazyflie_server-1]: started with pid [17202]
WARNING: Package "cfclient" does not follow the version conventions. It should not contain leading zeros (unless the number is 0).
process[crazyflie1/crazyflie_add-2]: started with pid [17203]
WARNING: Package "cfclient" does not follow the version conventions. It should not contain leading zeros (unless the number is 0).
process[crazyflie1/joy-3]: started with pid [17208]
[ INFO] [1435859830.461233524]: Opened joystick: /dev/input/js0. deadzone_: 0.050000.
WARNING: Package "cfclient" does not follow the version conventions. It should not contain leading zeros (unless the number is 0).
Traceback (most recent call last):

  File "/home/maverick/indigo_ws/src/crazyflie_ros/crazyflie/scripts/crazyflie_server.py", line 287, in <module>
    import cflib
ImportError: No module named cflib

process[crazyflie1/quadrotor_teleop-4]: started with pid [17236]
WARNING: Package "cfclient" does not follow the version conventions. It should not contain leading zeros (unless the number is 0).
process[crazyflie1/crazyflie_demo_controller-5]: started with pid [17266]
WARNING: Package "cfclient" does not follow the version conventions. It should not contain leading zeros (unless the number is 0).
[INFO] [WallTime: 1435859830.619038] wait_for_service /add_crazyflie
process[rviz-6]: started with pid [17279]
WARNING: Package "cfclient" does not follow the version conventions. It should not contain leading zeros (unless the number is 0).
process[rqt_plot-7]: started with pid [17282]
[crazyflie_server-1] process has died [pid 17202, exit code 1, cmd /home/maverick/indigo_ws/src/crazyflie_ros/crazyflie/scripts/crazyflie_server.py __name:=crazyflie_server __log:=/home/maverick/.ros/log/a31abbc2-20df-11e5-977f-d0df9a8cc6bf/crazyflie_server-1.log].
log file: /home/maverick/.ros/log/a31abbc2-20df-11e5-977f-d0df9a8cc6bf/crazyflie_server-1*.log
WARNING: Package "cfclient" does not follow the version conventions. It should not contain leading zeros (unless the number is 0).
process[rqt_plot2-8]: started with pid [17304]

whoenig, 2015-07-02:
Make sure that you either install the Bitcraze Python SDK or point to it using "roslaunch crazyflie_demo teleop_xbox360.launch crazyflieSDK:=/path/to/the/sdk"

Additionally, I am in the process of switching to the C++ version by default, which does not have any dependencies. If you want to try, take a look at the "cpp" branch.

rs1990, 2015-07-02:
Hello , I tried the cpp branch and while it compiled fine i get the following error. i can echo the crazyflie1/joy topic and i see msgs being publiched when i use the joystick. nothing else

NODES
  /crazyflie1/
    crazyflie_add (crazyflie/crazyflie_add.py)
    crazyflie_demo_controller (crazyflie_demo/controller.py)
    joy (joy/joy_node)
    quadrotor_teleop (hector_quadrotor_teleop/quadrotor_teleop)
  /
    crazyflie_server (crazyflie/crazyflie_server)
    rqt_plot (rqt_plot/rqt_plot)
    rqt_plot2 (rqt_plot/rqt_plot)
    rviz (rviz/rviz)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
WARNING: Package "cfclient" does not follow the version conventions. It should not contain leading zeros (unless the number is 0).

ERROR: cannot launch node of type [crazyflie/crazyflie_server]: can't locate node [crazyflie_server] in package [crazyflie]

WARNING: Package "cfclient" does not follow the version conventions. It should not contain leading zeros (unless the number is 0).
process[crazyflie1/crazyflie_add-2]: started with pid [27311]
WARNING: Package "cfclient" does not follow the version conventions. It should not contain leading zeros (unless the number is 0).
process[crazyflie1/joy-3]: started with pid [27312]
WARNING: Package "cfclient" does not follow the version conventions. It should not contain leading zeros (unless the number is 0).

rs1990, 2015-07-02:
I also get this error while running setup.sh while installing the crazyflie-clients-python.

copying lib/cflib/crtp/radiodriver.py -> build/lib.linux-i686-2.7/cflib/crtp
error: package directory 'lib/cfclient/utils/inputdevices' does not exist

whoenig, 2015-07-02:
Do you have the file "/home/maverick/indigo_ws/devel/lib/crazyflie/crazyflie_server"? If not, the compilation did not succeed fully.

Not sure about the other error - which version of the SDK are you using? I usually don't install and specify the path to the SDK on the command line (or change the launch files directly).

rs1990, 2015-07-02:
Hi. I got it working I guess it was an issue with my installation. ran it a couple of times and it got fixed. Btw  can i publish a Twist msg to get it to fly?

whoenig, 2015-07-02:
Awesome! I just switched the default to be the cpp version (now in master branch; older python version in python branch). There is also new documentation on how to use it with a joystick etc.
In principle it is just a Twist msg - checkout https://github.com/whoenig/crazyflie_ros#cmd_vel for details.

closed 2015-07-28 by whoenig