whoenig/crazyflie_ros #6: Missing AddCrazyflie.h

weiweikong, 2015-07-16:
Hi,
   just compiled these project under ROS indigo and got following report
```
/home/joker/crazyflie_ros/src/crazyflie_ros/crazyflie_driver/src/crazyflie_add.cpp:2:36: fatal error: crazyflie/AddCrazyflie.h: No such file or directory
 #include "crazyflie/AddCrazyflie.h"
                                    ^
compilation terminated.
/home/joker/crazyflie_ros/src/crazyflie_ros/crazyflie_driver/src/crazyflie_server.cpp:2:36: fatal error: crazyflie/AddCrazyflie.h: No such file or directory
 #include "crazyflie/AddCrazyflie.h"
                                    ^
compilation terminated.
make[2]: *** [crazyflie_ros/crazyflie_driver/CMakeFiles/crazyflie_add.dir/src/crazyflie_add.cpp.o] Error 1
make[1]: *** [crazyflie_ros/crazyflie_driver/CMakeFiles/crazyflie_add.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[2]: *** [crazyflie_ros/crazyflie_driver/CMakeFiles/crazyflie_server.dir/src/crazyflie_server.cpp.o] Error 1
make[1]: *** [crazyflie_ros/crazyflie_driver/CMakeFiles/crazyflie_server.dir/all] Error 2
make: *** [all] Error 2
```
  It seems `crazyflie/AddCrazyflie.h` is missing and I could not get this file in this project and Github. Any help about this issue ?

whoenig, 2015-07-16:
Hi, there was a missing dependency to the message generation. If you pull my latest changes it will hopefully work on your machine as well.

weiweikong, 2015-07-17:
Hi, thank you for your reply. I update the recent CMakelist, but the project still failed, after deleting the files in `build` folder.

```
Scanning dependencies of target _crazyflie_driver_generate_messages_check_deps_AddCrazyflie
Scanning dependencies of target std_msgs_generate_messages_py
Scanning dependencies of target std_msgs_generate_messages_lisp
Scanning dependencies of target _crazyflie_driver_generate_messages_check_deps_UpdateParams
Scanning dependencies of target std_msgs_generate_messages_cpp
Scanning dependencies of target crazyflie_controller
[  0%] [  0%] [  8%] Built target std_msgs_generate_messages_py
Built target std_msgs_generate_messages_cpp
Building CXX object crazyflie_ros/crazyflie_controller/CMakeFiles/crazyflie_controller.dir/src/controller.cpp.o
[  8%] Built target _crazyflie_driver_generate_messages_check_deps_AddCrazyflie
[  8%] Built target std_msgs_generate_messages_lisp
[  8%] Built target _crazyflie_driver_generate_messages_check_deps_UpdateParams
Scanning dependencies of target crazyflie_driver_generate_messages_cpp
Scanning dependencies of target crazyflie_driver_generate_messages_py
Scanning dependencies of target crazyflie_driver_generate_messages_lisp
[ 25%] Built target crazyflie_driver_generate_messages_cpp
[ 41%] Built target crazyflie_driver_generate_messages_lisp
Scanning dependencies of target crazyflie_add
Scanning dependencies of target crazyflie_server
[ 66%] Built target crazyflie_driver_generate_messages_py
[ 75%] Scanning dependencies of target crazyflie_driver_generate_messages
Building CXX object crazyflie_ros/crazyflie_driver/CMakeFiles/crazyflie_add.dir/src/crazyflie_add.cpp.o
[ 83%] [ 91%] [100%] [100%] Built target crazyflie_driver_generate_messages
Building CXX object crazyflie_ros/crazyflie_driver/CMakeFiles/crazyflie_server.dir/src/Crazyflie.cpp.o
Building CXX object crazyflie_ros/crazyflie_driver/CMakeFiles/crazyflie_server.dir/src/Crazyradio.cpp.o
Building CXX object crazyflie_ros/crazyflie_driver/CMakeFiles/crazyflie_server.dir/src/crazyflie_server.cpp.o
/home/amax/Workspace/crazyflie/src/crazyflie_ros/crazyflie_driver/src/crazyflie_server.cpp:2:36: fatal error: crazyflie/AddCrazyflie.h: No such file or directory
 #include "crazyflie/AddCrazyflie.h"
                                    ^
compilation terminated.
/home/amax/Workspace/crazyflie/src/crazyflie_ros/crazyflie_driver/src/crazyflie_add.cpp:2:36: fatal error: crazyflie/AddCrazyflie.h: No such file or directory
 #include "crazyflie/AddCrazyflie.h"
                                    ^
compilation terminated.
make[2]: *** [crazyflie_ros/crazyflie_driver/CMakeFiles/crazyflie_add.dir/src/crazyflie_add.cpp.o] Error 1
make[1]: *** [crazyflie_ros/crazyflie_driver/CMakeFiles/crazyflie_add.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[2]: *** [crazyflie_ros/crazyflie_driver/CMakeFiles/crazyflie_server.dir/src/crazyflie_server.cpp.o] Error 1
make[1]: *** [crazyflie_ros/crazyflie_driver/CMakeFiles/crazyflie_server.dir/all] Error 2
Linking CXX executable /home/amax/Workspace/crazyflie/devel/lib/crazyflie_controller/crazyflie_controller
[100%] Built target crazyflie_controller
make: *** [all] Error 2
Invoking "make -j24 -l24" failed
```

It seems that `crazyflie_driver_generate_messages` has been built, however, the `AddCrazyflie.h' is also needed. I also tried to comment this line, but it occurred more errors. In the `crazyflie_ros/crazyflie` directory, there are no cpp and h files. Is that correct ?

whoenig, 2015-07-17:
Hi, it looks like I forgot to do a clean build. The package crazyflie got renamed a few days ago to crazyflie_driver and catkin_make was looking in the wrong folder. It should be fixed now! Please let me know if that works for you as well. For your reference, AddCrazyflie.h should be generated in devel/include/crazyflie_driver/.

weiweikong, 2015-07-20:
Hi, it work fine! Thanks!

closed 2015-07-20 by weiweikong