whoenig/crazyflie_ros #7: How to set proper constellations for multi-UAV using Vicon system ?

weiweikong, 2015-07-20:
Due to the small size of Crazyflie, which kind of constellations for each Crazyflie with Vicon system are recommended ?

whoenig, 2015-07-20:
Do you mean constellations of reflective markers on the Crazyflie or constellations of Crazyflies in the air?

weiweikong, 2015-07-21:
I mean the reflective markers for Vicon system.

whoenig, 2015-07-21:
It depends on the Vicon setup. In general you need distinguishable configurations, where the markers are as far away as possible. The less accurate your space, the further away you need to place the markers. I use 6.4mm spheres on top of a custom 3D printed frame (there are pictures in the Bitcraze blog: https://www.bitcraze.io/2015/05/autonomous-flight-using-kinect2-for-position-control/). I put 3 markers on the frame and 2 markers on the Crazyflie itself. That way you can get at least 6 different configurations. If your Vicon system is less accurate, you might need to extend the arms of the quad to attach the markers further away.

weiweikong, 2015-07-21:
Great! I saw the detail of your setup. Thanks!

closed 2015-07-28 by whoenig

araujokth, 2015-12-23:
Hi Wolfgang, are you sharing the 3D printed frame files online? I was thinking of making the same frame as you have for my CFs! thanks a lot!