whoenig/crazyflie_ros #9: testing connection to crazyflie

lucas-t, 2015-07-31:
How do I test if the connection between the running services and Crazyflie (2.0) was successfull?

Steps I did: 
1.) roslaunch crazyflie_server.launch
2.) roslaunch crazyflie_add.launch

I also cannot find the 
"crazyflie_driver/AddCrazyflie.h" and "crazyflie_driver/UpdateParams.h"
which are included i.e. in the crazyflie_server.cpp.

-ROS: Jade
-Ubuntu: 14.04

Thanks for the support.

whoenig, 2015-08-01:
The package is designed to reduce the latency for possible reconnections. As such, it does not have the connected/disconnected events as the cfclient. It simply re-sends packets as fast as possible. That allows to have faster controller loops for the position controller.
The easiest way to test is to use a joystick and the launch files in the crazyflie_demo package (e.g. "roslaunch crazyflie_demo teleop_xbox360.launch". Furthermore, the LED on the Crazyradio blinks green if everything is working fine (and red if there are packet drops). I could add some kind of status message containing statistics about packet drop rate if that is of interest.

The header files you are looking for are generated during catkin_make from the message files. You can find them in your workspace folder under devel/include/crazyflie_driver/.

lucas-t, 2015-08-10:
Thank you for your quick reply.

closed 2015-11-02 by whoenig