whoenig/crazyflie_ros #10: Cmake Error

jakestmart, 2015-08-06:
Hi,

I have attempted to compile the ros file using the Jade distribution.

using clean make and catkin_make commands I still receive the same error.

I have looked at the cmakefillist on line 8 it has: find_library(USB_LIB usb-1.0)

I have also cloned the package and still receiving the same error.

Where can I locate the usb-1.0 library or ammend the cmakelist?

Error Returned:
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
USB_LIB
    linked by target "crazyflie_server" in directory /home/bitcraze/catkin_ws/src/crazyflie_ros/crazyflie_driver

-- Configuring incomplete, errors occurred!
See also "/home/bitcraze/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/bitcraze/catkin_ws/build/CMakeFiles/CMakeError.log".
Makefile:456: recipe for target 'cmake_check_build_system' failed
make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed

whoenig, 2015-08-06:
Try using `sudo apt-get install libusb-1.0-0-dev` to install libusb. This is required to communicate with the Crazyradio USB Dongle. Please let me know if it fixes your issue so that I can update the documentation accordingly.

jakestmart, 2015-08-07:
Apt-get has solved my problem thank you.

I have attempted to execute "roslaunch crazyflie_demo teleop_xbox360.launch uri:=radio://0/100/2M" but it returns and says that this is not a launch file in the package?

whoenig, 2015-08-10:
It sounds like your workspace is not in your path. From your workspace folder, execute `source devel/setup.bash`. Afterwards (same command prompt), use `roslaunch crazyflie_demo teleop_xbox360.launch uri:=radio://0/100/2M`. Technically it is not required to run roscore separately if you use roslaunch (it will automatically start a roscore if none is running). You can update your `~/.bashrc` to always include your workspace if you use only one.

whoenig, 2015-08-10:
Also, you don't need to run anything apart from `roslaunch crazyflie_demo ...`. In particular, the crazyflie_server is run automatically by the launch file.

jakestmart, 2015-08-11:
Thanks for the help! I've been able to get it to run and the demo as well. 

Is there a personal channel I can contact you on? As I will be using your ROS packages as part of my project and I'd like to acknowledge you.

whoenig, 2015-08-11:
I am glad you got it working! Feel free to contact me via E-Mail; You can find my contact information on the ros wikipage (http://wiki.ros.org/crazyflie) or in the `package.xml` files of any of the ROS packages I provided.

closed 2015-11-02 by whoenig