whoenig/crazyflie_ros #11: Best way to use Vicon?

4estlaine, 2015-09-08:
When using the Vicon system, what is the best way to position the markers? I'm having a hard time getting the minimum 3 markers on the crazyflie such that I can create a subject and observe it using the vicon bridge.

Is there a way to track just a single marker (and thus unlabeled?) I'm relatively new to the Vicon system.

4estlaine, 2015-09-08:
Nevermind. It turns out my cameras weren't properly calibrated. After calibrating, I was able to create an object by placing one 6mm static marker on top of each propeller.

whoenig, 2015-09-08:
You can use a single marker, but won't get 6D pose information (just 3D). That is mainly an issue for the controller (which needs to know yaw) and also very problematic in terms of noise from the vicon system.

For my setup you can refer to issue #7.

closed 2015-09-08 by whoenig