whoenig/crazyflie_ros #12: Catkin_make error -> Invoking "make -j1 - l1" failed

rshum19, 2015-09-08:
Hi, 

I'm trying to install the package in a Jade distribution.

I cloned the repository to ~/catkin_ws/src and called catkin_make. Midway through the process I get the error "Invoking "make -j1 -l1" failed".

Any inside on why I'm getting this error?

thanks

whoenig, 2015-09-10:
Which compiler are you using? I tested on Ubuntu 14.04 with Jade and it compiled without problem.

The issue is caused by a C++11 feature (http://en.cppreference.com/w/cpp/thread/sleep_for), but C++11 is enabled in the CMakeLists.txt (https://github.com/whoenig/crazyflie_ros/blob/master/crazyflie_driver/CMakeLists.txt#L13). So it looks like your compiler is ignoring the C++11 setting for some reason.

Also, please provide some more details on the compilation by using `catkin_make VERBOSE=1`. This should print out which arguments catkin used for the compiler to rule any Makefile-related issues out.

rshum19, 2015-09-10:
Hi, 

as you pointed out it was a compiler issue. I updated the compiler to the latest version of gcc and catkin_make compiled without error. 

However, now when I run roslaunch crazyflie_demo teleop_xbox360.launch uri:=radio://0/100/2M I get the following errors:
started core service [/rosout]
ERROR: cannot launch node of type [crazyflie_driver/crazyflie_server]: can't locate node [crazyflie_server] in package [crazyflie_driver]
ERROR: cannot launch node of type [crazyflie_driver/crazyflie_add]: can't locate node [crazyflie_add] in package [crazyflie_driver]
ERROR: cannot launch node of type [joy/joy_node]: joy

any inside on this?? All the help is appreciate

whoenig, 2015-09-10:
Did you register your workspace with ROS, i.e. did you run `source devel/setup.bash` in your workspace? (Or add a similar line to your `.bashrc`?). The other error about joy_node sounds like you did not install the package. Try `sudo apt-get install ros-jade-joy` to install it.

closed 2016-01-31 by whoenig