whoenig/crazyflie_ros #14: Error launching the demo files

swapnil1704, 2015-09-25:
I'm trying to roslaunch the demo flies but I'm getting 

[ERROR] [1443209803.433117116]: Exception thrown while processing service call: Could not find Crazyradio with given devId!
[ERROR] [1443209803.433242638]: Service call failed: service [/add_crazyflie] responded with an error: Could not find Crazyradio with given devId!

Can you help to solve the error.

whoenig, 2015-09-25:
Are you able to use the crazyflie python client as a normal user (i.e. without sudo). Or in other words, did you follow the instructions to make the CrazyRadio useable without being root (see https://github.com/bitcraze/crazyflie-clients-python#setting-udev-permissions)?

swapnil1704, 2015-10-02:
Thanks I'm able to run it now. I had not made CrazyRadio useable without being root.

closed 2015-10-02 by whoenig