whoenig/crazyflie_ros #15: Hover issues

jakestmart, 2015-10-02:
Continuining on from this reply:

"Your VRPN server should publish to "/crazyflie". "/crazyflie/base_link" is only required for visualization in RVIZ, and you can easily add a static transformation (see https://github.com/whoenig/crazyflie_ro ... launch#L40). 

You said you can take-off: Does that mean autonomously? That indicates that your optitrack works fine (otherwise it would just not take off). Are you using CF 1.0 or 2.0? Which firmware are you running?"

1. I have configured the server to publish to /crazyflie
2. Yes I can take off
3. Currently the crazyflie ramps to 50,000 then cuts out. - I have tried it with 3 different crazyflies and still the same issue
4. Using CF 1.0 - I am using the fimrware source from your github.

I have attached my frame configuration and graph of the ros environment.
[image]
[image]

whoenig, 2015-10-02:
At 3.: What does "cuts out" mean? The motors suddenly stop and it drops down?
I never tried the current hover controller with a CF 1.0, so there might be some tuning required to keep it stable. In particular, there is a YAML file with all the constants I am using: https://github.com/whoenig/crazyflie_ros/blob/master/crazyflie_controller/config/crazyflie2.yaml
(And there is no crazyflie1.yaml yet...)

Your frame and graph look fine to me.

jakestmart, 2015-10-04:
3. Yes, that's correct. It begins to hover, then drops.  I have modified the crazyflie2.yaml. 

I've used the standard constants supplied by the firmware. 

The Optitrack axis uses Y in the positive upward position - where as the crazyflie ros package uses Z as the upward axis.

Q1. Are the units of the coordinates in cm or m? 
Q2. I have swapped the outputs published by the vrpn to match the crazyflie ros ( Y upwards) including quanterions 
Q3. Is it possible to have a dynamic rate for the subscriber to tf and pose in controller.py? Ive noticed the vrpn frames average slightly vary around 120.5-121.5hz or am I on the wrong track ?

whoenig, 2015-10-04:
Q1, Q2: Everything follows the ROS standards, i.e. units are in meters, the coordinate system is right-handed with Z up (see http://wiki.ros.org/geometry/CoordinateFrameConventions for more details). You would need to make sure that the data you get from OptiTrack matches that (however, if you use an OptiTrack ROS package, this should take care of that for you.) Which VRPN/OptiTrack package are you using?
Q3: The controller always uses the latest frame, i.e. it doesn't really matter if there is a slight delay. In fact, by default the controller update frequency is set to just 50 Hz (this allows to get more CFs on a single radio; feel free to increase the number if you work with a single CF or use one radio per CF). It is a parameter for the controller, see https://github.com/whoenig/crazyflie_ros/blob/master/crazyflie_controller/src/controller.cpp#L248 for details. You could set it to 120 to get best performance - faster doesn't really help since your MoCap is not faster anyways.

jakestmart, 2015-10-08:
Q1. I am using the Vrpn Client Ros Indigo (http://wiki.ros.org/vrpn_client_ros) with Optitrack Motive: Tracker 1.8 (http://www.optitrack.com/products/motive/tracker/). 
Optitrack defines its coordinate system as:
Y_optitrack (up) "Yaw"
X_optitrack ( Across) "Pitch"
Z_optitrack ( Towards yourself) "Roll"

I've swapped the variables to match the ros system.
x_ros = Z_opitrack "roll"
y_ros = X_opitrack "pitch"
z_ros = Y_opitrack "yaw"

Q2. I've set the rate of the topic /crazyflie/goal to 120hz to match the vrpn rate.

Q4. I assume that controller.py changes the PID values for the crazyflie quadcopter OR its the PID values for the control parameters for the ROS environment pose estimation?

whoenig, 2015-10-08:
Q1: Optitrack uses a left-handed coordinate system, while ROS using a right-handed one. Hence, just swapping variables is not enough, you also need to swap the sign. Also make sure that you adjust the whole pose (i.e. position and orientation).

Q2: Sounds great!

Q4: The ROS driver does not touch the PID controller onboard. All the parameters are for the position controller (which is part of the ROS stack and a set of PIDs for x,y,z, and yaw.) You'll need to tune those even if your PIDs on-board are fine.

jakestmart, 2015-10-08:
Q1. I have alligned the frames to reflect opitrack 

Q4. Okay what is the purpose of and what do they do?
crazyflie2.yaml : I modified the PID parameters for the ros control here. 
i.e 'rostopic echo /crazyflie/controller/PIDs/x/kp' returns the .ymal kp,ki, kd values 

controller.py: The PID values here at https://github.com/whoenig/crazyflie_ros/blob/master/crazyflie_controller/scripts/controller.py 
From my understanding they are involved with the ros control?

whoenig, 2015-10-09:
Q1: Which format is the second file?

Q4: The onboard controller is an attitude controller. It takes thrust, roll, pitch, and yaw and makes sure that the CF reaches that pose. The outer (ROS) controller is a position controller, i.e. it takes a goal position (x,y,z,yaw) as input and computes the attitute (roll,pitch,yaw,thrust) as output. The current controller achieves that by using PIDs on x,y,z,yaw independently. This is by no means the best way to do it. Changing the yaml-file just updates that outer PID values. You would not need to update it for a CF2.0, but possibly for a CF1.0.

controller.py is not used anymore (I should go ahead and delete that). Everything is in C++. Make sure that your launch file uses the C++ controller (see https://github.com/whoenig/crazyflie_ros/blob/master/crazyflie_demo/launch/hover_vicon.launch#L28-L30). It looks like the vrpn-version of that launch file was not updated accordingly.

jakestmart, 2015-10-09:
Q1: 
>Second file is the echo from /tf within the ROS environment.
> This diagram explains how I've attained my values: https://www.dropbox.com/s/073w546dxrx2k0u/Optitrack%20to%20ROS%20frames.pdf?dl=0
> I've changed the values in the vrpn_client_node.cpp (https://github.com/clearpathrobotics/vrpn_client_ros/blob/indigo-devel/src/vrpn_client_ros.cpp)

      tracker->pose_msg_.pose.position.x = tracker_pose.pos[1]*-1;
      tracker->pose_msg_.pose.position.y = tracker_pose.pos[0];
      tracker->pose_msg_.pose.position.z = tracker_pose.pos[2];

     tracker->pose_msg_.pose.orientation.x = tracker_pose.quat[1]*-1;
      tracker->pose_msg_.pose.orientation.y = tracker_pose.quat[0];
      tracker->pose_msg_.pose.orientation.z = tracker_pose.quat[2];
      tracker->pose_msg_.pose.orientation.w = tracker_pose.quat[3];

Note: I've changed it for angular and linear accel too.

Q4. I've checked my launch file it does use the crazyflie2.launch file

whoenig, 2015-10-10:
Q1: Great! I think that is the correct approach.

Could you add some debug output to print the transform:
https://github.com/whoenig/crazyflie_ros/blob/master/crazyflie_controller/src/controller.cpp#L169
and also use `rostopic echo /crazyflie/cmd_vel` to see what the output is? It looks to me that for some reason once the controller switches to "Automatic" mode the thrust suddenly becomes 0. This could be either by a wrong transform or wrong PID values.

jakestmart, 2015-10-10:
Unfortunately I have spent about 3 hours to print the transform. Is there a simple line that you could provide me? 
ROS_INFO just prints out where it is unfortuantely I've used ROS_DEBUG but it won't let me print the transform.
Main console: https://www.dropbox.com/s/lzskkkguq82d8uv/main_console_log.txt?dl=0
rostopic echo cmd_vel: https://www.dropbox.com/s/q7sdime7hpgzxqj/cmd_vel_log.txt?dl=0

Notes:
1. When Optitrack cannot track the crazyflie it remains in automatic mode - Should it jump to "emergency" when it isn't tracked?

Is it possible if we could perhaps Skype, I am in Australia Melbourne? - It would be great to get this system working considering how far I've gone. This project is basically overdue, so any help would be greatly appreciated!

whoenig, 2015-10-10:
I don't think there is a out-of-the-box way to print it. You can use the following (put it after the `lookupTransform` call):
`ROS_INFO("Is: [%f,%f,%f], Goal: [%f,%f,%f]", transform.getOrigin().getX(), transform.getOrigin().getY(), transform.getOrigin().getZ(), m_goal.pose.position.x, m_goal.pose.position.y, m_goal.pose.position.z)`

Technically we are only interested in the z-position since the motors simply seem to shut down once you enter the Automatic mode. Your cmd_vel topic output is strange - the z component constantly switches between the minimum (10000) and maximum (60000) value. That either means the units are wrong or that you get weird outlier data from your motion capture system.

The controller assumes that the Mocap is "mostly" working, i.e. it takes the latest transformation. If there are longer periods of none-tracking it simply won't work. I didn't make the attempt to automatically go into emergency mode in that case. It would be an option for the future though. In general you should get an updated pose at least about 30 times per second for reasonable performance. You can use your joystick to enter emergency mode manually (red cross on an XBOX controller).

I don't think that skype will help much here. At the end I will need to look at the data you are providing me and that is easier/more efficient using github. I am on Pacific Standard Time.

jakestmart, 2015-10-11:
Okay So I have the current output taken from the transform: https://www.dropbox.com/s/89833prpl2dtk9p/transform_log.txt?dl=0

I have looked at the units and echo'd a direct coordinate from optitrack below:
- Optitrack: https://www.dropbox.com/s/8vjbr2mjimbhqpg/optitrack_coordiantes.png?dl=0
- Ros system: https://www.dropbox.com/s/u0wohlxswfo0jcb/tf_echo_transform.png?dl=0

I noticed maybe the quanterions don't match even when I have converted them from degrees to radians.
I.e Assuming my switching of axis is correct from the diagram above.
let Z_ros = Y_optitrack
Z_ros = 0.0229471mm ~ Y_vrpn = 22.9191 mm

Z_ros_ ROTATION= 0.012567 rads    
Y_vrpn_ORIENTATION = -86.455 degrees  ==  -1.508924405 rads

Which is unusual ..  I think this is where the problem is. 
From the Vrpn_client my current configuration is:
tracker->pose_msg_.pose.position.x = tracker_pose.pos[1]*-1;
  tracker->pose_msg_.pose.position.y = tracker_pose.pos[0];
  tracker->pose_msg_.pose.position.z = tracker_pose.pos[2];

 tracker->pose_msg_.pose.orientation.x = tracker_pose.quat[1]*-1;
  tracker->pose_msg_.pose.orientation.y = tracker_pose.quat[0];
  tracker->pose_msg_.pose.orientation.z = tracker_pose.quat[2];
  tracker->pose_msg_.pose.orientation.w = tracker_pose.quat[3];

whoenig, 2015-10-11:
For the first log: can you include the cmd_vel output as well (print it using ROS_INFO too)? So far it seems very strange on all axis. For the z-component you would expect that it should reach its goal (but it doesn't). For the x/y it seems to diverge rather than converge...

The easiest way to test your transformation is to visualize it in RVIZ. If you use the launch files I provided, there should be already RVIZ visualization support. Just don't press the take-off button on the joystick, but move the Crazyflie manually and see if RVIZ shows the movement (translation and rotation) correctly. Your code seems correct to me, but there could be other issues with the approach (such as a bug in the VRPN optitracker server).

Even if there is an issue with the rotation, the Crazyflie should still reach its goal height as the z-controller works independently of any rotation.

jakestmart, 2015-10-11:
Okay so this is the updated log: https://www.dropbox.com/s/nh02i4ax7lful4j/transform_cmd_vel_log.txt?dl=0

I've looked at the translations they are correct but the rotations dont seem to match up which is puzzling me...

whoenig, 2015-10-11:
Great! What was the issue?

For x/y I would leave ki 0 (or close to 0). Try reducing kp. You can also reduce the min/max allowed angles to avoid fast movements.

You only need to change thrust if there is an issue with height stabilization. Typically the height is not very stable initially, but gets better over time after the controller adjusted the integral portion.

Try to fly higher - at least 0.5, better 0.7m. Otherwise you do get ground-effects (if you hover at 0.3m manually using the joystick you'll see what I mean).

jakestmart, 2015-10-11:
The issue was:
1. Accuracy / calibration of Optitrack system - I did a full recalibration 
2. Changed the positon of the markers so that the propelers do not interfere 

Currently you have a high Kd value within the .yaml file, I was under the impression that we do not need KD

Thank you for that, will try the 0.5-0.7 and see how that goes and will tune it.

jakestmart, 2015-10-13:
@whoenig I have modified the demo.py script to fit for stampedtransform
Edited demo.py: https://www.dropbox.com/s/4uij5n9z9k2nyp9/demo.py?dl=0
Edited demo1.py: https://www.dropbox.com/s/ze3zfytb30vfn6v/demo1.py?dl=0
Edited: hover_vrpn_waypoint.launch: https://www.dropbox.com/s/2tpb8hy99vptu4q/hover_vrpn_waypoint.launch?dl=0

let me know what you think? I will try these scripts tomorrow.

jakestmart, 2015-10-15:
I am going to create another thread since there is a different issue

closed 2015-10-15 by jakestmart

Akash-Pardasani, 2016-12-08:
Hi @jakestmart @whoenig I was trying to hover a CF 2.0 and as mentioned above I used hover_vrpn_waypoint.lauch and made all the changes as well. I'm getting errors regarding use of demo1.py (Below are the specified errors)
[image]

whoenig, 2016-12-08:
Which branch is this? demo.py on the master branch doesn't even have line 53 (https://github.com/whoenig/crazyflie_ros/blob/master/crazyflie_demo/scripts/demo.py).

Akash-Pardasani, 2016-12-08:
Hi @whoenig ! Thank you for the quick reply. I'm using the modified demo.py and demo1.py uploaded by @jakestmart . Although I got rid of some indentation errors in demo.py .
But now the CF won't remain in flight for long after the takeoff command(It hovers and then drops by itself).
I've introduced the changes in vrpn_client_ros.cpp as mentioned above and highlighted in the given file. 
[vrpn_client_ros.pdf](https://github.com/whoenig/crazyflie_ros/files/640351/vrpn_client_ros.pdf)

The CF works fine with cfclient and I've done recalibration twice as well.
Any help would be appreciated.

whoenig, 2016-12-08:
Does normal takeoff+hover work (i.e. without the use of the demo script)? Also, you should be able to rule out issues with your VRPN changes just by looking at RVIZ and checking if the shown position matches ROS standards by moving it around manually (without flying).

s3xyxavy, 2018-02-12:
Hi @jakestmart understand that you edited the vrpn_client_ros.cpp , to tracker_pose.pos[1]*-1 .. etc I am not able to run that config, but more of i kept running the old one even though I have changed. I am very new to ROS, May I know what are steps taken to reconfig to this new setup?
Also does ur 
tracker->twist_msg_.twist.angular.x = pitch
tracker->twist_msg_.twist.angular.y = yaw
tracker->twist_msg_.twist.angular.z = roll
Look like that?