whoenig/crazyflie_ros #16: Flying 3 Crazyflie 1.0 Quads

jakestmart, 2015-10-15:
1. Multi Hover with 3 quads - I am able to connect to two, but the system cannot connect to the third one. I've check if its an issue witht the quad so instead I switched the sequence of which quad connects first. It always lets the first two connect.
> I have flashed all 3 CF1.0 to the same address: //0/10/2M/'address' - I don't think I can bootload configure the 'address' value for CF1.0
Q1. How would I go about running one ROS environment for 2 x Crazyradio dongles? i.e 6 CF 1.0 on 2 dongles
This is my launch file for multi hover: https://www.dropbox.com/s/3ma6s5hqvfkkdx4/multi_hover_vrpn_3.launch?dl=0

2. Way point Navigation 
@whoenig I have modified the demo.py script to fit for stampedtransform
Edited demo.py: https://www.dropbox.com/s/4uij5n9z9k2nyp9/demo.py?dl=0
Edited demo1.py: https://www.dropbox.com/s/ze3zfytb30vfn6v/demo1.py?dl=0
Edited: hover_vrpn_waypoint.launch: https://www.dropbox.com/s/2tpb8hy99vptu4q/hover_vrpn_waypoint.launch?dl=0

let me know what you think? I will try these scripts tomorrow.

jakestmart, 2015-10-15:
I noticed that I Have to make changes to add for the address in the stm32 firmware. But CF1.0 doesnt have an nrf chip.. 

https://www.bitcraze.io/2015/04/towards-crazyswarms-2/

whoenig, 2015-10-16:
1. CF1.0 does not support the addressing yet, as far as I know. It is fairly simply to implement though (it requires changes to the STM32 firmware and the python client). The changes are extremely similar to the ones I did back when I implemented it for CF2.0. The reason I didn't fully support it was that I don't have multiple CF1.0. (It is possible to mix CF1.0 with CF2.0 though).

Unfortunately, without that support you won't be able to fly multiple of them reliably.  

1b.) You simply change the address to "//1/10/2M/address" -- the first number is the id of your dongle. The ids are sequential starting from 0. I.e. for 6, you duplicate whatever you have and just update the dongle-id accordingly.

2. It should be certainly StampedTransform rather than transform only. Thank you for the hint!

whoenig, 2016-01-31:
The issues with StampedTransform have been fixed in the master branch a while back.

closed 2016-01-31 by whoenig