whoenig/crazyflie_ros #17: Using Optitrack instead of Vicon

rshum19, 2015-10-21:
Hi

first of all great work, thank you for sharing! 

I'm trying to get hovering working with an Optitrack Mocap system instead of a Vicon system. However, I'm new to ROS so I'm having some trouble.

I'm trying to run the hovering example you have. 
I modified hover_vicon.launch file as follow
              <include file="$(find vicon_bridge)/launch/vicon.launch"/>  
for
              <include file="$(find mocap_optitrack)/launch/mocap.launch"/>

to launch the optitrack system instead of the vicon system. I'm using the "Mocap_Optitrack" package to retrieve the CF position. (http://wiki.ros.org/mocap_optitrack). This seems to work fine.

I think the issue is providing the correct node to hover_vicon.launch. The optitrack package can publish poses of configured rigid bodies as tf transforms, poses, and/or 2D poses. 

However, I don't know if this are compatible with your controller code or that needs to be modified too. Can you give me some pointers?

Thnak you

whoenig, 2015-10-21:
Hi,
there are people using Optitrack with my controller and there is no limitation in that regard. All you need is the tf-frame-name for the Crazyflie. The code right now assumes that the base-frame is called "world" - if that is not the case, you have to either adjust the code or create a static transformation between your base frame and world.

I am personally not familiar with the mocap_optitrack package. I think most other people used VRPN (http://wiki.ros.org/vrpn_client_ros). However, in that case you need to flip the axis: The crazyflie driver assumes ROS standards, which is different from what Optitrack sends out. Please read issue #15 for some more details on that.

Best,
Wolfgang

rshum19, 2015-10-21:
HI Wolfgang, 

Thank you for the rapid response. 

1) My base-frame is called "world" so I think that is not the issue.
on the argument   "frame:= <t>", should <t> be the tf-frame-name?

In my case I have 2 topics the publish different data which one is the correct one if any?

/crazyflie/pose
https://goo.gl/photos/fyMqRGcYxk6gPnzS8

/tf
https://goo.gl/photos/N9UoakBHPJcQkZQ98

2)  Also when running "roslauch crazyflie_demo hover_vicon.launch ....." I get eh following log message, do you know what this is due to??

<Don't know ack: Port: 15 Channel: 3 Len: 1>

thanks
Roberto

whoenig, 2015-10-22:
1) You should use the tf system (not messages). You can look at the available frames using `rosrun tf view_frames` (and look at http://wiki.ros.org/tf/Tutorials/Introduction%20to%20tf for more details).
2) You might be running a newer firmware than I tested with. It essentially means that the ROS node could not interpret one of the results from the CF. Nothing worrisome though - Port15 is the link layer (https://wiki.bitcraze.io/projects:crazyflie:firmware:comm_protocol) and has no relevance for ROS.

rshum19, 2015-10-24:
Thank you Wolfgang that helped and solved the problem I was having. However, now I noticed that the controller is not doing anything. The rostopic /crazyflie/cmd_vel is always 0 and the crazyflie does not anything. Do you have any possible insight to why this could be? Sorry that I have no more info to describe the problem besides its not working

whoenig, 2015-10-24:
Are you giving it the take-off command using a joystick (or calling the service manually on the command line)? Once it gets the command the motors should at least ramp up - only afterwards the controller is actually becoming active.

rshum19, 2015-10-24:
Yeah my joystick was not configured properly. Thanks for the help

closed 2015-11-02 by whoenig

gotozzj, 2016-08-19:
Hi whoenig 

I'm trying to hover the crazyflie. I've modified hover_vicon.launch like what rshum19 did, since I'm using optitrack as well. 
But I have some problems to make the crazyflie take-off. 
1. I don't have a joystick, but I don't know how to give take-off command.
2. I realized that the axis of optitrack and the axis of ros are different, which file do I need to modify? 
sorry for my bad English.

whoenig, 2016-08-22:
1. Takeoff is just a ROS service. You can use the command line or write a small script which calls the service (see https://github.com/whoenig/crazyflie_ros/blob/master/crazyflie_demo/scripts/controller.py as an example on how you could do it using python.)
2. It is best to make optitrack consistent with the ROS standard, rather than fixing it on the Crazyflie side (otherwise you will have issues with virtually all ROS packages.) You can try doing the conversion in vrpn_client_ros (or whatever node you use to connect OptiTrack to ROS.)

Akash-Pardasani, 2016-12-09:
@rshum19 Could you please elaborate on how you configured the joystick ?
My console doesn't take any commands further than this even after giving input from the joystick. 
[image]
The CF works fine with cfclient and RViz display is fine as well.

whoenig, 2016-12-09:
You can test your joystick by looking at the "/joy" topic. You'll always need to do additional operations on another terminal - the roslaunch terminal won't allow additional user-input unless you shut down ROS.

byronxubin, 2017-02-26:
Hi @whoenig 

I am using optitrack and I think I have configured everything, which could be shown both in the output of position and visualization in Rviz. But the issue is that when I execute the take off command in another terminal, the CF just flying around and have no idea where the position I gave. By the way, I found that the cmd_val topic, there is linear velocity in x,y which drove the CF to fly away from the target. I am quite confused at this consequence. What might be wrong here? Thanks in advance.

whoenig, 2017-02-26:
It sounds to me like your coordinate system does not match ROS standards. If there would be no transformation at all, the CF would not even take off. Please verify your coordinate system, as OptiTrack uses a different one. It is also important that the rotation is correct.

byronxubin, 2017-02-26:
Right now the problem is that the in the Rviz and the rostopic, the outputs all seem right to me. So how can I test if the coordinate is right? According to Motive and the `rostopic echo /optitrack/rigid_body`, I think they are compatible with each other, which should imply that the transform I apply is right. But my concern lies in `tf.broadcaster`, I'm not sure if I broadcast the right transformations.

miaoyanan, 2017-03-23:
Hi， @whoenig ，
I have some questions on Optitrack and ROS .I can fly one CF using the OPtitrack at the guide of yours, and now I want to fly multiple CFS.But there are something wrong with them ,when I set up the second rigid body at the motive app,it give the prompt :.

1. REQUIRED process [mocap_node -2]has died! process has died[pid 3619,exit code -11, ].
2.[mocap_node -2]killing on exit.[rosout-1]killing on exit
3.shuting down processing monitor.....shuting down processing monitor complete...done
Would you please give some adivces on solving it?Thanks in advance.

[image]

whoenig, 2017-03-23:
This doesn't seem to be related to crazyflie_ros. It looks like your "mocap_node" is getting a segfault. You should reach out for help from the maintainer of mocap_node.

miaoyanan, 2017-03-24:
But I can set up one rigid body ,using the same mocap system, without questions.Only when I set up  the second one ,is there something wrong.So I think it seems not the question of the mocap_node.Do you have another ideas?Thanks in advance.

BruceWheaton, 2017-03-24:
I have seen people talk about the OptiTrack protocol changing, and nodes crashing when more than one rigid body is used. Hopefully that's enough for you to google on?

miaoyanan, 2017-03-24:
really? Thank you .but I use the old version,I can find the answers  on the google,thank you!

miaoyanan, 2017-03-29:
Hi, @whoenig 
I have some troubles in flying.I communicate my CF by using  the mocop_optitrack1.8.1 .I use your code and change the hover_vicon to mocop_optitrack.But now I  change the parameters of kp、kd，ki of the Z oriention,the CF can fly well, but sometimes ,either there is offset in x,y oriention for flying  or the CF fly to  anywhere and can't to hover at set point.I don't know the reasons are caused by communication problems or the problems  for anastomosing controller algorithmm of x,y,z or other aspects.
I am looking forward to your reply!Thanks in advance.

whoenig, 2017-03-31:
The default gains for the controller should work. Monitor your position estimate as visualized in RVIZ. If there are communication issues there should be warnings. In general, if you are able to fly using joystick teleoperation, the communication should be "good enough" for the autonomous mode as well.

miaoyanan, 2017-04-03:
hello,
Now I want to take yaw control,how can I get the value of yaw by optitrack like the x-value which is gotten in the code (transform.getOrigin().x()).Please show me the real value of yaw.Thanks in advance!

whoenig, 2017-04-04:
Please ask basic questions about ROS on answers.ros.org

miaoyanan, 2017-04-05:
Hi,
How can I get the real value of yaw in crazyflie_controller.cpp,I write some code as followed,but it also some error, (/home/psy/crazyflie_ws/src/crazyflie_ros/crazyflie_controller/src/controller.cpp: In member function ‘void Controller::iteration(const ros::TimerEvent&)’:
/home/psy/crazyflie_ws/src/crazyflie_ros/crazyflie_controller/src/controller.cpp:235:28: error: expected primary-expression before ‘(’ token
               tf::Matrix3x3(  
                            ^
/home/psy/crazyflie_ws/src/crazyflie_ros/crazyflie_controller/src/controller.cpp:237:28: error: expected primary-expression before ‘(’ token
              tf::Quaternion( 
                            ^
/home/psy/crazyflie_ws/src/crazyflie_ros/crazyflie_controller/src/controller.cpp:242:16: error: expected primary-expression before ‘)’ token
                )
                ^
make[2]: *** [crazyflie_ros/crazyflie_controller/CMakeFiles/crazyflie_controller.dir/src/controller.cpp.o] 错误 1
make[1]: *** [crazyflie_ros/crazyflie_controller/CMakeFiles/crazyflie_controller.dir/all] 错误 2
make: *** [all] 错误 2
Invoking "make -j4 -l4" failed
)
can you give some help?(  My code as follows:     
                tf::StampedTransform transform;
                m_listener.lookupTransform(m_worldFrame, m_frame, ros::Time(0), transform);
                tfScalar roll, pitch, yaw;
               tf::Matrix3x3(
                  tf::Quaternion( 
                      transform.getRotation().x(),
                     transform.getRotation().y(),
                     transform.getRotation().z(),
                    transform.getRotation().w(),
                 )).getRPY(roll, pitch, yaw);)

miaoyanan, 2017-04-05:
Hi,
I have another question about the crazyflie.When I launch the file ,there are also some error.
{ [ INFO] [1491392637.450594245]: Elapsed: 0.564156 s
[INFO] [WallTime: 1491392637.479644] found update_params service
[INFO] [WallTime: 1491392637.480471] waiting for emergency service
[INFO] [WallTime: 1491392637.485088] found emergency service
[INFO] [WallTime: 1491392637.486046] waiting for land service
terminate called after throwing an instance of 'tf2::LookupException'
  what():  "world" passed to lookupTransform argument target_frame does not exist. 
[crazyflie/controller-6] process has died [pid 25367, exit code -6, cmd /home/psy/crazyflie_ws/devel/lib/crazyflie_controller/crazyflie_controller __name:=controller __log:=/home/psy/.ros/log/255fd77e-19f5-11e7-9e71-68f728de612d/crazyflie-controller-6.log].
log file: /home/psy/.ros/log/255fd77e-19f5-11e7-9e71-68f728de612d/crazyflie-controller-6*.log
}
I don't know how to solve these and wwhere to ask this kind of question,please give me some advice. I am looking forward to your reply.Thank you very much.

miaoyanan, 2017-04-05:
Hi ,@whoenig 
Sometimes,when I launch the hover.launch ,there are some warning issues.whether is it the communication issues,and how can I solve this. Thank you very much!

byronxubin, 2017-04-05:
@miaoyanan 
There is one error in your code. You have one more parentheses after 'getRPY()'. I guess it should be your problem.
`tf::StampedTransform transform;
m_listener.lookupTransform(m_worldFrame, m_frame, ros::Time(0), transform);
tfScalar roll, pitch, yaw;
tf::Matrix3x3(
tf::Quaternion(
transform.getRotation().x(),
transform.getRotation().y(),
transform.getRotation().z(),
transform.getRotation().w(),
)).getRPY(roll, pitch, yaw);)`

miaoyanan, 2017-04-17:
Hello, @whoenig 
When I hover my CF using the optitrack system,sometime either there is offset in x,y oriention for flying or the CF fly to anywhere and can't to hover at set point.I also update the latest firmware,but It can't be solved.Can you tell me how can I solve the problem.
  Thank you very much!Best wishes to you !

whoenig, 2017-04-19:
Make sure your OptiTrack tracking matches the ROS coordinate system. You can find more details in our book chapter "Flying Multiple UAVs Using ROS", preprint available at http://usc-actlab.github.io/publications.html.

neudinglei, 2017-05-03:
Hello, @whoenig @rshum19 @byronxubin 
Now,I want to do collaborative research with my colleagues by the Crazyflie2.0. I am using the optitrack system and ros package mocap_optitrack.The problem is that it can not communicate with three or more Crazyflie2.0。Can you give me some advice and solutions? Also, you can tell me some collaborative control studies and cases by the Crazyflie2.0。Thank you very much!

whoenig, 2017-05-04:
There is some more documentation here for setup and usage: http://crazyswarm.readthedocs.io/en/latest/. Research using the Crazyflie platform is listed here: https://www.bitcraze.io/research/

neudinglei, 2017-05-08:
@whoenig 
Hello! whoenig
Thank you for your reply. I still have communication problems.
The system is optitrack. Now  the ros package is vrpn_client_ros.When I run the launch file (roslaunch crazyflie_demo hover_vrpn.launch) on a terminal, it did not suggest any errors.But, it can not get the data from optitrack(using motive on windows PC),that is to say that the actual location of the crazyflie can not be obtained.Can you provide a solution?
In the hover_vrpn.launch, The parameters are set as follows；
<arg name="ip" default="192.168.56.1" />（ip of windows PC）
<arg name="port" default="3883" />
<arg name="uri" default="radio://0/80/250K/E7E7E7E7E7" />
<arg name="frame" default="/world>
Is there a problem with parameter settings?Hope to receive your reply。thank you!

whoenig, 2017-05-08:
Frame should match the object name in OptiTrack Motive. Otherwise I suggest trying vrpn_client_ros independently of crazyflie_ros and if it doesn't work seek out help for that package or from OptiTrack.

miaoyanan, 2017-07-31:
Hi,@whoenig 
I can fly mutiple crazyflie2.0 by using one crazyradio.but there are some warning about comunicated link.Now I want use two crazyradio to control the different cfs at the same time,how can I set the crazyradio
and how can I identify which CF is controlled by which crazyradio.
   Thank you in advance！

whoenig, 2017-08-01:
the first number in the uri is the radio number. So just plug in two radios, and use `radio://0/...` and `radio://1/...` as URIs. You won't know which Crazyradio is which one, but that shouldn't really matter in practice anyways.

neudinglei, 2017-08-03:
@whoenig 
Hello whoenig!
I have recently done more unmanned aerial vehicles in collaborative control experiments and have encountered some questions that you can help.
First describe a phenomenon: with a radio control three crazyfile 2.0 fixed-point flight,  when the distance between UAVs is small (for example, less than 1m), the UAV will be in shock. But once the interval between unmanned aerial vehicles increases, the shock will be reduced a lot. (Single UAV fixed point control, the effect is very good). With a radio control two crazyfile 2.0 fixed-point flight, even if the interval is less than 0.3m, almost no shock. Can you explain the above?
In addition, I have a few questions about radio:
1. Are there frequent warnings of connection quality that affect the control of the aircraft?
2. a radio control several aircraft best?
3. How does a computer communicate with multiple aircraft through two radio?
I hope to hear from you soon, thank you！

neudinglei, 2017-08-03:
@whoenig 
There is a problem in controlling UAVs: when the actual position is in error with the desired position, the output of the position controller is not 0,which is 1 or 2 or less, the UAV's attitude is not adjusted. In other words, when the position control output posture set value is relatively small, the UAV's attitude controller is not adjusted, guess is due to the attitude controller plus dead zone, please give confirmation, thank you!

neudinglei, 2017-08-03:
I am using your ros package to control the Crazyflie 2.0.I have a question：
you know: the attitude setting is passed to the aircraft via radio. Since the aircraft is sent to the attitude setting, why does the underlying code of the aircraft have positional control?
Hope you can reply as soon as possible, thank you！

neudinglei, 2017-08-07:
@whoenig 
There is a problem with the control of the UAV: When the actual position is deviated from the desired position, the output of the position controller is not 0, 1 or less, but the attitude of the UAV is not adjusted. In other words, when the position controller output the attitude setting is small, UAV posture is not adjusted, what is the reason? Under normal circumstances, the attitude set value is not 0, the aircraft's attitude should be adjusted until achieve the desired position. Hope to reply as soon as possible, thank you!

whoenig, 2017-08-08:
You might need to adjust the PID gains (https://github.com/whoenig/crazyflie_ros/blob/master/crazyflie_controller/config/crazyflie2.yaml).

miaoyanan, 2018-01-11:
Hi,@whoenig 
There is a problem with my control of my cf.When I use the x_box360 to control the cf,It can communicate and control normally. when I use the mocap_optitrack,I can get the data .however I can't communicate with cf by crazy_radio.I don't  know the reason,can you help me?
Thank you in advance!

whoenig, 2018-01-11:
@miaoyanan Please open a new issue. This issue has been closed. In the new issue, add some more details about your setup, i.e. which launch file are you using? What does "I can get the data" mean (which data etc.)? How do you know that you "can't communicate" in that case?