whoenig/crazyflie_ros #18: PID Controller query

rshum19, 2015-10-31:
Hi Whoening, 

I'm trying to understand your code a little better and I have a doubt in your PID controller I was hoping you could clear it out for me. In lines 192-195 in controller.cpp (https://github.com/whoenig/crazyflie_ros/blob/master/crazyflie_controller/src/controller.cpp#L192-195)
                msg.linear.x = m_pidX.update(0, targetDrone.pose.position.x);
                msg.linear.y = m_pidY.update(0.0, targetDrone.pose.position.y);
                msg.linear.z = m_pidZ.update(0.0, targetDrone.pose.position.z);
                msg.angular.z = m_pidYaw.update(0.0, yaw);

why do you send a 0 instead of the current drone position from the mocap to the PID controller input argument value? 

Thanks

whoenig, 2015-11-01:
In line 180 the target position is transformed into the crazyflie coordinate system. If the two (crazyflie absolute position and target position) are the same, we are done (and the values are 0 in that case.) You can also try to do it the other way around, but then you would need to take care of the coordinate transformation for the control signal (e.g. if your yaw is 90 degrees, you will need to fly forward in order to reach a goal which is sideways in absolute coordinates.) Transforming the goal into the CF-frame takes care of the transformation implicitly.

rshum19, 2015-11-15:
Ok that make sense.  As a follow up question in your approach when you initialize the mocap system your CF onboard frame has to be aligned with the frame from the mocap system? This the yaw that board sensors read matches the yaw of the mocap system correct?

whoenig, 2015-11-19:
That actually doesn't matter. The input for the Crazyflie is just yaw-rate, not absolute yaw (that is also because the CF does actually not use the onboard magnetometer to get an absolute value.) Given that, the PID controller will just send appropriate yawrate commands to the CF to reach the target-yaw value.

closed 2016-01-31 by whoenig