whoenig/crazyflie_ros #20: multi waypoint not working

rs1990, 2015-12-04:
Hello,

Whenever i try to launch the multi_waypoint_vicon demo, i keep having this issue

[ WARN] [1449246837.823356858]: grab frame returned false
[ INFO] [1449246837.824576985]: creating new object crazyflie1/crazyflie1 ...
[ WARN] [1449246837.827497608]: unable to load zero pose for crazyflie1/crazyflie1
[ INFO] [1449246837.827772853]: ... done, advertised as " vicon/crazyflie1/crazyflie1" 
[WARN] [WallTime: 1449246837.944989] Could not process inbound connection: topic types do not match: [geometry_msgs/PoseStamped] vs. [geometry_msgs/Pose]{'topic': '/crazyflie1/goal', 'tcp_nodelay': '0', 'md5sum': 'd3812c3cbc69362b77dc0b19b345f8f5', 'type': 'geometry_msgs/PoseStamped', 'callerid': '/crazyflie1/controller'}

i changed the launch file to make it run for just one crazyflie. Any help

Thanks

whoenig, 2015-12-06:
I just pushed a change which should fix your issue. Please verify and let me know!

rs1990, 2015-12-15:
Hello,

I don't get that error and the program runs. But for some reason it just starts to fly and in 2-3 seconds it stops and doesn't continue. The 2 crazyflies are not able to take off more than 1-2 inches above the ground. I don't get any errors. Im not sure whats wrong. 

Any idea ? I really am in a bind and would appreciate any help.

Thank you,

whoenig, 2015-12-15:
Does it work with a single Crazyflie (i.e. `hover_vicon.launch`)?

rs1990, 2015-12-15:
Hello,

Yes , I can get single and multiple(2) hover working. But run into problems on multi waypoint (single and double).

I tried using a single crazyradio for both as well as a two radios for both. It just jumps for a second nd lands.

whoenig, 2015-12-15:
I just pushed another change to the launch file which hopefully fixes the issue you are seeing. Note that I don't have access to a motion capture space right now to try it out though. Please let me know how it goes!

rs1990, 2015-12-15:
Sure,

I will check it out tomorrow morning and get back to you. What do you
suspect the issue is ?.

Thank you,

whoenig, 2015-12-15:
This launch file was just not updated in a while. I changed the way on how to use the controller slightly and the launch file was not updated. At the end, it didn't load the PID parameters from the YAML file, causing the sudden drop after the take-off sequence.

rs1990, 2015-12-16:
Hello,

The new launch file worked great. Thanks a lot. 

There is one small error. With respect to the frame of crazyflie :

    <include file="$(find crazyflie_controller)/launch/crazyflie2.launch">
      <arg name="frame" value="$(arg frame)" />
    </include>

 -> Arg frame should be arg frame1.

Thank you so much for all your help.

Cheers

whoenig, 2015-12-16:
Thanks for the feedback. I fixed those arguments!

closed 2015-12-16 by whoenig