whoenig/crazyflie_ros #21: connection from PC to crazyflie not working?

araujokth, 2015-12-14:
Hi,

I am trying to run the demo example "roslaunch crazyflie_demo teleop_xbox360.launch" and I get the data correctly from the crazyflie (battery, temperature, IMU) but for some reason I cannot send commands to the crazyflie using my logitech gamepad. I tried the cfclient and it works to receive data and control the crazyflie with the gamepad, but for some reason it is not working with your demo example. I tried also to use the logitech_gamepad.launch from the hector project (had to modify it as you did for the xbox360.launch, but also did not work). Any good way to debug this? 

I am running ROS Jade, Ubuntu 14.04, Crazyflie 2.0.

Thanks for the help!

whoenig, 2015-12-14:
Once you run `roslaunch crazyflie_demo teleop_xbox360.launch`, can you check `rostopic echo /crazyflie/joy` and see if you get any messages from the joystick back? Also, while you are doing that you can take a note of the axis and button mapping (just check which values change). If the rostopic doesn't publish, you might need to use another joystick device (e.g. `/dev/input/js1`).

araujokth, 2015-12-15:
Thanks so much for the quick reply. I was getting an error output on "joy" but since the program was continuing to print I could not see that yesterday. I had to install ros-jade-joy that for some reason it had not been installed. Now everything works! Thanks a lot for the help! I will be trying to modify your vicon demo for the qualisys mocap system during the day!

whoenig, 2015-12-15:
For some reason ros-jade-joy is not part of the default ros installation. Glad you figured it out!

closed 2015-12-15 by whoenig