whoenig/crazyflie_ros #22: Using Qualisys mocap

araujokth, 2015-12-15:
Hi,

I am trying to change the vicon mocap to the qualisys mocap and run the hover demo. I am using the ROS driver provided by https://github.com/KumarRobotics/motion_capture_system.  I get the following errors when running the demo code inside RViz

* Fixed Frame [world] does not exist
* No transform from [crazyflie/base_link] to [world] 

which is because I am not having a world frame output. When performing rosrun tf tf_monitor the output I get is (one of the continuous prints)

Frames:
Frame: /crazyflie/base_link published by unknown_publisher Average Delay: -0.0997525 Max Delay: 0
Frame: crazyflie published by unknown_publisher Average Delay: 0.00022675 Max Delay: 0.00036054

I have very new to both the mocap and ROS and I am not able to understand what is meant by the world frame and how I get to output that. I read the issue #17 but I could not make sense of this part. 

Also, what would the input to "frame" be in my case when running  "roslaunch crazyflie_demo hover_qualisys.launch uri:=radio://0/80/2M frame:=crazyflie x:=0 y:=0 z:=1"

Thanks for the help again!

araujokth, 2015-12-15:
I think I have solved this but now that I was going to try it the battery died, so I have to confirm after the battery is charged. 

I just had to rename "mocap" by "world" as "<param name="fixed_frame_id" value="world"/>" in the file qualisys.launch and now no error is shown in rviz.

I am also running "roslaunch crazyflie_demo hover_qualisys.launch uri:=radio://0/80/2M frame:=crazyflie x:=0 y:=0 z:=1"

whoenig, 2015-12-15:
Great that you got it working! Currently, the package assumes that the global frame is named "world". If that is not the case, you'll need to add a static transformation between "mocap" and "world".

closed 2015-12-15 by whoenig