whoenig/crazyflie_ros #26: Control with protective frame

araujokth, 2016-01-21:
Hi Wolfgang,

When you started flying with the frame, did you have to make some adjustments to the crazyflie (roll and pitch trimming, internal PID controller) or the position controller in ROS? 

I was just testing some configurations on a 3D printed frame I found online http://www.thingiverse.com/thing:1151252 and I noticed that the crazyflie became much more sensitive, and several of the crazyflies I tested just started oscillating when holding a position and some became unstable. I then tested a specific configuration in two different crazyflies and noticed that in one there were almost no oscillations and in the other there were a lot, which makes me think that I really need to adjust the propellers very well, or that there may be some differences with respect to the IMU, so that one of them would need roll/pitch trimmings and the other didn't. One thing I noticed is that even if the markers are very lightweight, their configuration must be symmetric, otherwise the crazyflie tilts a lot and becomes unbalanced.

Did these issues also happened to you when using a frame? It would be great to know if you have any tips regarding this.

Thanks a lot for the help again!

Best,
José

whoenig, 2016-01-21:
I didn't change anything on-board (i.e. no roll/pitch trimming, no change of the internal PID controller). The ROS controller, however, was only tuned with the frame on (otherwise I wouldn't have been able to get the position feedback.) I did balance my propellers (however I have the first gen of Crazyflie2 which had some issues with the first batch of rotors.)

I certainly see a change in the agility due to the added payload, but not to the extend you describe. What is the added payload and what kind of markers are you using? I tried 6.4mm markers and they work fine, but of course that depends on your motion capture as well.

Isn't the wireless charger on the bottom of the CF? So I am confused how that changes the battery placement? Furthermore, I would try reducing any additional payload. If I have markers and LED deck on the Crazyflie I do see it becoming slightly unstable as well. Your total payload should certainly not exceed 15g, and less is better.

Best,

Wolfgang

araujokth, 2016-01-22:
Hi Wolfgang,

thanks for the tips!

Things got improved today after balancing the propellers better. However, I still get quite some oscillating movement and mostly feel this when I am landing. 

Today I looked more closely to your youtube video of the swarm experiment and noticed that your CFs also oscillate quite a bit around the waypoint, which is similar to the behaviour I get. In my previous tests I hot glued the markers to the CF frame so I managed to do position tracking without the frame and the CF was super stable and none of this oscillatory behaviour occurred. So perhaps its really an issue that comes by having a not completely balanced marker pattern + frame + slightly unbalanced propellers.

I checked the weight and we are at 40g total with the frame, qi charger and markers, instead of 28g with just the battery and the battery holder, so our payload is 12g. I am using 10 mm markers, but the weight difference is not very big. I am trying to get a hold of the 7mm markers to try it instead.

The large oscillations could also be because my 3D printed frame is not very aerodynamic. Compared to yours, mine has straight edges, which is probably not so good. Also, my frame is not symmetric in both axis, as yours. In case you would be able to share your frame, it would be great to try if there is any difference between the one I am using and yours.

I will try two things now in the weekend, one is to try to adjust the position tracking controlling and see if the oscillations get reduced. I tried a bit of that yesterday but did not help a lot though. The other thing I will try is to setup a frame with carbon fiber sticks assembled in the center of the CF, and see if the oscillations disappear. If thats the case, I would have to rethink of using the 3D printed frame. 

Another thing I noticed is that the attitude PID controller from the newest firmware has slightly different parameters from the one that is in your git repository. I will also try to change these parameters and try.

Thanks again for all the help!
.

whoenig, 2016-01-25:
Hi José,

The oscillations in the video only happen with a swarm of Crazyflies - a single one is pretty stable. I am not exactly sure what the reason is, but it is a combination of motion capture issues, latency issues (radio is shared between 2 CFs), and additional disturbances created by the additional airflow.

I didn't make the frame myself and am still in the process with the author to determine an appropriate license. Until then, feel free to contact me via E-Mail to get an STL-file.

Another option would be to replace the motor-mounts with custom ones. Bitcraze published OpenSCAD models here for 3D-printable ones (https://github.com/bitcraze/bitcraze-mechanics/tree/master/cf2-mount-openscad). I could image that extending the model with an "outward-facing stick" to glue the marker on could help a lot. Btw: I use Command Poster strips to glue markers on the frame, which is very sturdy, light-weight, and still not permanent (can remove without breaking frame and/or marker!). If you do end up getting better marker holders, please do let me know!

Also try with reduced payload, and see if that has n influence like I noticed. If that is the case you might be able to upgrade the motors to slightly larger ones to still use all your payload.

Thank you for keeping me up-to-date!

Wolfgang

araujokth, 2016-01-26:
Hi again,

sorry for the late reply! 

I managed to re-tune the external controller to make the CF very stable with the frame, but unfortunately I lost 1 minute of flying time, I think because of the added weight. Because of this, I bought carbon fiber sticks and hot glued them to the battery holder and things work very well and the sticks are quite stable. I attached three markers with double-side tape to three of the motor holders, and then attached two to two carbon fiber sticks.

Thanks again for the help and I will keep you updated!

closed 2016-01-26 by araujokth

byronxubin, 2017-02-23:
Hi, José

I am right now using the same frame as you mentioned above. Similarly, I am facing the same problem as you described. Mine is even worse, I have only tested on one CF right now. The CF would be unstable nine out of ten times. What I observed is that the motion capture system loses tracking quickly sometimes. So I am wondering if you can share how you place your markers. Thanks a lot.

Bin

araujokth, 2017-02-23:
Hi, you can find some images of the placements here https://www.flickr.com/photos/ericssonuxlab/sets/72157663659656919. I basically put some markers with hot glue on each edge, and then some upwards or to the side using carbon fiber sticks. That worked very well in most of the configurations. I would try to make it as symmetric as possible. As you will see in the images, I have many asymmetric configurations because I had to have 15 different patterns as I was flying 15 at the same time so ran out of symmetric configurations that are distinguishable quite fast

byronxubin, 2017-02-24:
Hi, thank you for your quick reply. This gives me clue for how to place the markers. Flying 15 CFs at the same time is quite awesome. By the way, what is the mocap system are you using? I am using optitrack and I am using this package right now. `https://github.com/crigroup/optitrack` Although I have modiefied a little bit in this package to make it compatible with ROS, but it seems that it not works quite well. Thanks a lot.

araujokth, 2017-02-24:
Thats strange it doesnt work well. I would contact them if thats the case :) I used the Qualisys system which works great and I highly recommend http://www.qualisys.com/

byronxubin, 2017-02-24:
Oh. I thought you were using optitrack. Right now we only have optitrack in the lab. So I should stick to that, no choice. Right now I am trying to make new crazyflie frame to test, if it would have better performance. Thanks anyway.

whoenig, 2017-02-26:
You can use VRPN with OptiTrack if the dedicated optitrack package does not work well for you. If you are interested in larger swarms, I recommend taking a look at the Crazyswarm project, which has been shown to work with up to 49 CFs and identical marker configurations.

byronxubin, 2017-02-26:
If the dedicated optitrack package still does not work, I think I would consider choosing VRPN. I have searched VRPN and vrpn_client_ros, but I have no idea about how to use it. Since VRPN acts like a server, so do I need to install it on the windows along with OptiTrack software Motive?