whoenig/crazyflie_ros #28: Hovering using VICON problem

OliverFaust, 2016-02-05:
Hi,
Excellent contribution. 

I am new to Vicon, so the following might be quite trivial to solve. 

The Crazyflie_tools and the teleop_xbox360 demo are working. However, I am struggling with the hover_vicon demo. To be specific, when I do:
roslaunch crazyflie_demo hover_vicon.launch uri:=radio://0/80/250k frame:=/vicon/crazyflie/crazyflie x:=0 y:=0 z:=1
There are no errors in the console log, but the RViz GUI shows two errors:
1) Fixed Frame Fixed Frame [world] does not exist
2) crazyflie/base_link No transform from [crazyflie/base_link] to [world]

Any pointers on what might be wrong?

Thank you for your help in advance.

whoenig, 2016-02-05:
Try testing `vicon_bridge` separately: `rosrun vicon_bridge vicon_bridge`. Now look at your tf-tree: `rosrun tf view_frames`. It should generate a pdf with the transformation hierarchy. There should be a world frame and a vicon/crazyflie frame as well.

OliverFaust, 2016-02-08:
Hi,
Thank you for the quick response. 

However, just 
`rosrun vicon_bridge vicon_bridge`
followed by 
`rosrun tf view_frames`
results in a pdf with one node: no tf data received.

In contrast, if I do
`roslaunch crazyflie_demo hover_vicon.launch uri:=radio://0/80/250k frame:=/vicon/crazyflie/crazyflie x:=0 y:=0 z:=1`
The pdf contains two nodes, vicon/crazyflie/crazyflie and crazyflie/base_link. They communicate:
> Broadcaster: /crazyflie/baselink_broadcaster
> Average rate: 10.165 Hz
> Most recent transform: 1454924509.741 ( -0.027 sec old)
> Buffer length: 4.919 sec
But there is no world frame and no vicon/crazyflie frame.

whoenig, 2016-02-08:
That means your `vicon_bridge` is not working properly. The nodes you see in the second case is a static transformation to show the 3D model in rviz properly - they don't have any further value.

This can have multiple causes:
 * Communication issues. Make sure you have the Tracker software running on your other machine and that you can ping hostname `vicon` from your ROS-computer. (If your hostname is not vicon follow the instructions from the wiki: http://wiki.ros.org/vicon_bridge)
 * Make sure you have at least one object active in Vicon Tracker -- ideally named `crazyflie`
 * Watch out for any errors or warning when you run `vicon_bridge`

closed 2016-03-16 by whoenig