whoenig/crazyflie_ros #30: Exception thrown while processing service call: Could not find Crazyradio with given devId!

rs1990, 2016-02-17:
Hello,

I recently reformatted my system and got back to work but i cannot seem to get the crazyflies to connect. It was working befor but now its giving me this error

````
process[crazyflie/controller-6]: started with pid [32333]
[ INFO] [1455740151.123687176]: Adding radio://0/80/2M as crazyflie with trim(0.000000, 0.000000). Logging: 1, Parameters: 0
LIBUSB_ERROR_ACCESS
[ERROR] [1455740151.131435717]: Exception thrown while processing service call: Could not find Crazyradio with given devId!
[ERROR] [1455740151.131795260]: Service call failed: service [/add_crazyflie] responded with an error: Could not find Crazyradio with given devId!
process[crazyflie/pose-7]: started with pid [32353]
/vicon/crazyflie/crazyflie
````

I tried it on several systems and got the same results. I do not have the crazyflie client installed but i dont think i need that to use this package.

Im connected to the vicon system and getting data and everything is as it should be.

Any help ?

Thanks.

whoenig, 2016-02-17:
Did you set the udev permission for the Crazyradio (https://github.com/bitcraze/crazyflie-clients-python#setting-udev-permissions)?

rs1990, 2016-02-17:
Hello,

Yes I had made an error while copying. Sorry for the trouble.

Thanks for your help.

closed 2016-02-17 by whoenig

josephinemonica, 2016-09-05:
@whoenig Hi, I faced the same issue. I already set the udev permission. Could you help?

josephinemonica, 2016-09-05:
@whoenig Do I need to connect the radio and crazyflie using the pc client first before executing the demo?

whoenig, 2016-09-05:
Try if it works if you connect with the bitcraze python client (as normal user). Also, you need to reboot or reload the udev rules once you added that udev file.
You do not need the PC client (other than for such debugging purposes).

josephinemonica, 2016-09-05:
@whoenig it works now. Btw I'm a bit confused. When I run the teleop_ps3 demo,what does it suppose to do? Is it I can control the simulation using ps3 joystick, or I can control the actual crazyflie?

Cause I cant control both.
The drone does not appear in the simulation.
And the actual drone also won't move.

Thanks

whoenig, 2016-09-05:
I guess by "simulation" you mean rviz? That just shows the current state of the robot. Ideally, you would see the CF move if you use the joystick. If you use an actual PS3, make sure you press the PS3 button first (otherwise it won't send any joystick command). You can use `rostopic echo /crazyflie/cmd_vel` to see if there is anything != 0 send to the CF.

josephinemonica, 2016-09-05:
@whoenig I can't see anything except grid in the rviz

whoenig, 2016-09-05:
Check `rostopic list` and use `rostopic echo` on a few topics to make sure you get data.

josephinemonica, 2016-09-05:
@whoenig I checked rostopic echo /crazyflie/cmd_vel , and yes there is data sent, but the crazyflie doesn't move

whoenig, 2016-09-05:
Are there any other topics starting with `/crazyflie`? There should be an imu topic etc.

josephinemonica, 2016-09-05:
@whoenig I just realized that ususally when the crazyflie is connected with the radio, led M4, but it is not now

josephinemonica, 2016-09-05:
@whoenig yes there is /clicked_point
/crazyflie/battery
/crazyflie/cmd_vel
/crazyflie/imu
/crazyflie/joy
/crazyflie/magnetic_field
/crazyflie/pressure
/crazyflie/rssi
/crazyflie/temperature
/diagnostics
/initialpose
/move_base_simple/goal
/rosout
/rosout_agg
/tf
/tf_static

josephinemonica, 2016-09-05:
@whoenig I echo the battery, and imu, but nothing printed. Only the cmd_vel printed when I moved the joystick

whoenig, 2016-09-05:
Are you sure the uri you use is correct? I believe the master branch does not report an error - it will just wait forever until the CF responses if the uri is wrong.
You can try the requestBatching branch which has improved error handling (including timeouts for long-lasting requests).

josephinemonica, 2016-09-05:
@whoenig I'm sorry, I'm quite new. What do you mean by "using the correct uri"?

Also, does M4 led suppose to blink if it's connected to the radio? PS: The radio led is blinkng, but M4 on the Crazyflie is not

josephinemonica, 2016-09-05:
@whoenig How to check that the crazyflie is connected with the radio dongle? When I roslaunch, this is what appears (and I think it's fine) `ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[crazyflie_server-1]: started with pid [28210]
process[crazyflie/crazyflie_add-2]: started with pid [28211]
process[crazyflie/joy-3]: started with pid [28218]
process[crazyflie/quadrotor_teleop-4]: started with pid [28232]
process[crazyflie/crazyflie_demo_controller-5]: started with pid [28245]
process[rviz-6]: started with pid [28258]
process[rqt_plot2-7]: started with pid [28264]
process[rqt_plot3-8]: started with pid [28267]
[ INFO] [1473054242.086653502]: Opened joystick: /dev/input/js0. deadzone_: 0.050000.
[ INFO] [1473054242.134596594]: wait_for_service /add_crazyflie
[ INFO] [1473054242.139967021]: found /add_crazyflie
[ INFO] [1473054242.186836528]: Adding radio://0/80/2M as crazyflie with trim(0.000000, 0.000000). Logging: 1, Parameters: 1, Use ROS time: 1
Configured Dongle with version 0.53
[crazyflie/crazyflie_add-2] process has finished cleanly
log file: /home/osboxes/.ros/log/ac5c4842-7325-11e6-b552-000c29a1f7dd/crazyflie-crazyflie_add-2*.log
[INFO] [WallTime: 1473054242.937812] found update_params service
[INFO] [WallTime: 1473054242.938226] waiting for emergency service
[INFO] [WallTime: 1473054242.946057] found emergency service
`

whoenig, 2016-09-05:
The uri is the address of your Crazyflie (as shown in the combobox if you use the python client after you hit scan). You can either update the launchfile, or run it using `roslaunch crazyflie_demo teleop_ps3.launch uri:=radio://0/80/2M". You can also find the uri by running `rosrun crazyflie_tools scan`.
Not sure which one is M4 right now, but one of them should blink for exchanged packets.

whoenig, 2016-09-05:
There are additional printouts if it is connected successfully. In the master branch there is no clean way to find out. For proper error handling switch to the requestBatching branch (should be quite stable, I've been using it in the past.)

josephinemonica, 2016-09-05:
@whoenig when I scan, this is what I receive : LIBUSB_ERROR_BUSY
Could not find Crazyradio with given devId!

whoenig, 2016-09-05:
Make sure nothing else is using the radio.

josephinemonica, 2016-09-05:
@whoenig When I scan before roslaunch, I got :
 Configured Dongle with version 0.53
Receive LIBUSB_ERROR_TIMEOUT
Receive LIBUSB_ERROR_TIMEOUT
radio://0/100/2M

whoenig, 2016-09-05:
So your uri is different from the default. Use `roslaunch crazyflie_demo teleop_ps3.launch uri:=radio://0/100/2M`.

josephinemonica, 2016-09-05:
@whoenig THANK YOU so much. It works now. I didn't know about the uri before. Thank you

kashishdhal, 2020-03-31:
Hello @whoenig,

I am able to successfully launch hover_vicon.launch (please see below) as specified on your github. However, when I echo /crazyflie/imu or battery or pressure, I don't see any data. I made sure to source devel/setup.bash

Please let me know if you would be able to help me with the same. Note, I don't have a joystick but I don't know if that will make a difference in imu or pressure topics.

<pre><font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~</b></font>$ roslaunch crazyflie_demo hover_vicon.launch x:=0 y:=0 z:=0.5
... logging to /home/gnclab/.ros/log/40dd63d2-72fc-11ea-a41b-1c697a0794e3/roslaunch-system76-pc-16294.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
<font color="#CC0000">WARNING: disk usage in log directory [/home/gnclab/.ros/log] is over 1GB.</font>
<font color="#CC0000">It&apos;s recommended that you use the &apos;rosclean&apos; command.</font>

/opt/ros/melodic/lib/python2.7/dist-packages/roslaunch/loader.py:412: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  data = yaml.load(text)
<font color="#FCE94F">xacro.py is deprecated; please use xacro instead</font>
<b>started roslaunch server http://system76-pc:34579/</b>

SUMMARY
========

PARAMETERS
 * /crazyflie/controller/PIDs/X/integratorMax: 0.1
 * /crazyflie/controller/PIDs/X/integratorMin: -0.1
 * /crazyflie/controller/PIDs/X/kd: 20.0
 * /crazyflie/controller/PIDs/X/ki: 2.0
 * /crazyflie/controller/PIDs/X/kp: 40.0
 * /crazyflie/controller/PIDs/X/maxOutput: 10.0
 * /crazyflie/controller/PIDs/X/minOutput: -10.0
 * /crazyflie/controller/PIDs/Y/integratorMax: 0.1
 * /crazyflie/controller/PIDs/Y/integratorMin: -0.1
 * /crazyflie/controller/PIDs/Y/kd: -20.0
 * /crazyflie/controller/PIDs/Y/ki: -2.0
 * /crazyflie/controller/PIDs/Y/kp: -40.0
 * /crazyflie/controller/PIDs/Y/maxOutput: 10.0
 * /crazyflie/controller/PIDs/Y/minOutput: -10.0
 * /crazyflie/controller/PIDs/Yaw/integratorMax: 0.0
 * /crazyflie/controller/PIDs/Yaw/integratorMin: 0.0
 * /crazyflie/controller/PIDs/Yaw/kd: -20.0
 * /crazyflie/controller/PIDs/Yaw/ki: 0.0
 * /crazyflie/controller/PIDs/Yaw/kp: -200.0
 * /crazyflie/controller/PIDs/Yaw/maxOutput: 200.0
 * /crazyflie/controller/PIDs/Yaw/minOutput: -200.0
 * /crazyflie/controller/PIDs/Z/integratorMax: 1000.0
 * /crazyflie/controller/PIDs/Z/integratorMin: -1000.0
 * /crazyflie/controller/PIDs/Z/kd: 6000.0
 * /crazyflie/controller/PIDs/Z/ki: 3500.0
 * /crazyflie/controller/PIDs/Z/kp: 5000.0
 * /crazyflie/controller/PIDs/Z/maxOutput: 60000.0
 * /crazyflie/controller/PIDs/Z/minOutput: 10000.0
 * /crazyflie/controller/frame: /vicon/gncQuad/gn...
 * /crazyflie/controller/worldFrame: world
 * /crazyflie/crazyflie_add/enable_logging: False
 * /crazyflie/crazyflie_add/enable_logging_battery: True
 * /crazyflie/crazyflie_add/enable_logging_imu: True
 * /crazyflie/crazyflie_add/enable_logging_magnetic_field: True
 * /crazyflie/crazyflie_add/enable_logging_packets: True
 * /crazyflie/crazyflie_add/enable_logging_pose: False
 * /crazyflie/crazyflie_add/enable_logging_pressure: True
 * /crazyflie/crazyflie_add/enable_logging_temperature: True
 * /crazyflie/crazyflie_add/pitch_trim: 0
 * /crazyflie/crazyflie_add/roll_trim: 0
 * /crazyflie/crazyflie_add/tf_prefix: crazyflie
 * /crazyflie/crazyflie_add/uri: radio://0/80/2M
 * /crazyflie/crazyflie_add/use_ros_time: True
 * /crazyflie/joy/dev: /dev/input/js0
 * /crazyflie/joystick_controller/use_crazyflie_controller: True
 * /crazyflie/pose/name: goal
 * /crazyflie/pose/rate: 30
 * /crazyflie/pose/x: 0
[... 61 lines cut ...]
[ INFO] [1585623834.367665112]: Found variables in cache.
<b>[crazyflie/crazyflie_add-3] process has finished cleanly</b>
<b>log file: /home/gnclab/.ros/log/40dd63d2-72fc-11ea-a41b-1c697a0794e3/crazyflie-crazyflie_add-3*.log</b>
[ INFO] [1585623834.663208415]: Requesting memories...
[ INFO] [1585623834.700935269]: Memories: 8
[ INFO] [1585623834.739220622]: Ready...
[ INFO] [1585623834.739249062]: Elapsed: 0.475502 s
[INFO] [1585623834.818719]: found update_params service
[INFO] [1585623834.819596]: waiting for emergency service
[INFO] [1585623834.821553]: found emergency service
[INFO] [1585623834.822548]: waiting for land service
[INFO] [1585623834.824954]: found land service
[INFO] [1585623834.825984]: waiting for takeoff service
[INFO] [1585623834.828023]: found takeoff service
[ INFO] [1585623835.267563366]: ... connected!
[ INFO] [1585623835.267649953]: Setting Stream Mode to ClientPull: Success
[ INFO] [1585623835.267715562]: Axis Mapping: X-Forward Y-Left Z-Up
[ INFO] [1585623835.267798269]: Version: 1.3.0
[ INFO] [1585623835.267864272]: setting up grab_vicon_pose service server ... 
[ INFO] [1585623835.269357908]: setting up segment calibration service server ... 
<font color="#C4A000">[ WARN] [1585623835.273258602]: grab frame returned false</font>
<font color="#C4A000">[ WARN] [1585623835.277218383]: gncQuad occluded, not publishing... </font>
[ INFO] [1585623835.288111332]: creating new object gncQuad/gncQuad ...
<font color="#C4A000">[ WARN] [1585623835.290802863]: unable to load zero pose for gncQuad/gncQuad</font>
[ INFO] [1585623835.290888505]: ... done, advertised as &quot; vicon/gncQuad/gncQuad&quot;</pre>