whoenig/crazyflie_ros #32: How to properly configure Crazyflies for multi_teleop

raultron, 2016-05-09:
Hi,

Thank you very much for your package it has been really helpful.

I have four Crazfyflies 2.0 and two crazyradio dongles.

What I have done until now:

1. Installed the latest firmware in all the Crazyflies using the official client: [https://github.com/bitcraze/crazyflie-release/releases/tag/2016.02](https://github.com/bitcraze/crazyflie-release/releases/tag/2016.02)
2. Using the official client I modified the address, channel and datarate as follows:
    * Crazyflie 1: radio://0/120/2M/E7E7E7E700
    * Crazyflie 2: radio://0/120/2M/E7E7E7E701
    * Crazyflie 3: radio://1/100/2M/E7E7E7E702
    * Crazyflie 4: radio://1/100/2M/E7E7E7E703
3. I tested each crazyflie independently using: ```roslaunch crazyflie_demo teleop_xbox360.launch uri:=radio://0/120/2M```
4. Then I wanted to control two at the same time using: 
```roslaunch crazyflie_demo multi_teleop_xbox360.launch uri1:=radio://0/120/2M/0xE7E7E7E700 uri2:=radio//0/120/2M/0xE7E7E7E701```

The problem is that nothing happens, all the windows open (rviz, rqtplot) but no data is being received and no data is being sent, also there are no errors.

Then I tried to connect with the official client with the first pair of crazyflies turned on. Independent of the address value that I set on the client, both crazyflies react to the joystick and also I see a degradation of performance on the received values. My guess is that the firmware somehow is not implementing the addressing and since both crazyflies are using the same channel and datarate then both receive the joystick commands without addressing filtering. I also tried your old [client](https://github.com/whoenig/crazyflie-clients-python/tree/radio_multilink) with same results.

Can you please point me in the right direction?. Is there any special firmware which I have to install on the crazyflies?

Best regards,

Raul

raultron, 2016-05-09:
Nevermind... I solved it. In only flashed the firmware of the stm32 and forgot to flash as well the NRF chip, now I flashed the firmware of the nrf as well: [https://github.com/bitcraze/crazyflie2-nrf-firmware](https://github.com/bitcraze/crazyflie2-nrf-firmware) and it is working.

I will close this issue then :). Thanks anyway.

Raul

closed 2016-05-09 by raultron

byronxubin, 2017-04-04:
Hi raultron,

Do I need to flash crazyflie2-nrf-firmware to crazyflie if I want to use one crazyradio to connect two crazyflies? Thanks

ataffanel, 2017-04-04:
If you want to change the radio address of a Crazyflie 2.0 you might have to upgrade its firmware since the firmware flashed from factory did not have this functionality in the first batch. The latest [release](https://github.com/bitcraze/crazyflie-release/releases) will work.