whoenig/crazyflie_ros #33: Question about waypoint tracking controller

liwangGT, 2016-05-11:
Hi Whoening,

I am starting to work on crazyflie 2.0 for swarm robotics, and run into your ROS driver. It is pretty well organized. Thanks for your great work. I have a question about your waypoint tracking controller in "crazyflie_controller/src/controller.cpp" line 195-200.

` 
                geometry_msgs::Twist msg;
                msg.linear.x = m_pidX.update(0, targetDrone.pose.position.x);
                msg.linear.y = m_pidY.update(0.0, targetDrone.pose.position.y);
                msg.linear.z = m_pidZ.update(0.0, targetDrone.pose.position.z);
                msg.angular.z = m_pidYaw.update(0.0, yaw);
                m_pubNav.publish(msg);
`

The PID controller computes outputs using the current value and target value. However, why you use [0, 0, 0, 0] for the current value? Shouldn't you use the values updated from the tracking system?
`               position, quaternion = self.listener.lookupTransform(self.worldFrame, self.frame, t);
                rpy = tf.transformations.euler_from_quaternion(quaternion);
                msg.linear.x = m_pidX.update(position[0], targetDrone.pose.position.x);
                msg.linear.y = m_pidY.update(position[1], targetDrone.pose.position.y);
                msg.linear.z = m_pidZ.update(position[2], targetDrone.pose.position.z);
                msg.angular.z = m_pidYaw.update(rpy[2], yaw); `

Thanks,

Li

whoenig, 2016-05-11:
Hi Li,

The controller works in the body frame of the Crazyflie. In that frame, the current position is [0,0,0,0] and the goal pose is the one transformed from the world frame to the body frame.

Hope that helps,

Wolfgang

liwangGT, 2016-05-11:
Thanks for the quick reply. That explains my doubts.

Li

closed 2016-05-11 by liwangGT