whoenig/crazyflie_ros #34: crazyflie 2 doesn't respond to /cmd_vel topic

jhabhargav26, 2016-05-30:
Hi! 
I have connected the crazyflie 2.0 using crazyradio. I get all the data from IMU, magnetometer etc. But when I publish the values on /cmd_vel, crazyflie doesn't respond. I checked the topic using rostopic echo and it is getting the values. I have tried changing the radio, crazyflie, computer but the problem persists.
Could you please help me with this?

Thanks!
Bhargav Jha

whoenig, 2016-05-30:
It should work with the latest official firmware. One thing which was added a while ago in the firmware is called thrust lock. In this case, you need to send a few zero cmd_vel messages before actually sending what you want. This corresponds to joystick teleoperation, where you don't want the copter to start immediately after connection if by accident the joystick is not in zero pose. 
All the examples in the package (hover etc.) should account for that though and work out-of-the-box.

jhabhargav26, 2016-05-30:
Thanks Whoenig!
This solved my problem. I started with the non-zero value initially.

closed 2016-05-30 by jhabhargav26