whoenig/crazyflie_ros #36: Crazyflie won't log battery, temperature, etc.

spohorec, 2016-07-07:
IMU and RSSI are working fine, but I can't seem to get battery data through the package. I tried connecting via cfclient, and it showed me the battery level, but when I run`rostopic echo /crazyflie/battery`, there is no data returned. I haven't changed any of the default logging parameters, which are appearing as 'true'. Any idea why this might be happening?

spohorec, 2016-07-11:
Found the issue. baro.temp and baro.pressure don't seem to exist in the Crazyflie's parameters anymore. This was causing the "Size Doesn't Match!" error from Crazyflie.h and was causing it to not publish any of the parameters in the log2 block. I switched baro.temp and baro.pressure to other parameters since I didn't need them anyway, but this should perhaps be given a more robust solution.

whoenig, 2016-07-15:
Reopen so that I can fix that. I think this might be related to newer firmware versions.

whoenig, 2018-01-26:
Tested with the latest firmware and it works fine. Closing.

closed 2018-01-26 by whoenig