whoenig/crazyflie_ros #38: Syntax error in the controller?

antonioadaldo, 2016-08-23:
When

`rosrun crazyflie_controller controller.cpp`

I get

`line 9: syntax error near unexpected token `(''
`line 9: `double get(''

whoenig, 2016-08-23:
You can only rosrun executables, not source files. Try 
<code>rosrun crazyflie_controller controller</code>

antonioadaldo, 2016-08-23:
When
```
rosrun crazyflie_controller controller
```

I get

```
[rosrun] Couldn't find executable named controller below /home/adaldo/catkin_ws/src/crazyflie_ros/crazyflie_controller
```

(To compile my project I `catkin build` and then I `source /devel/setup.bash`.)

However, I found out that if I

```
roslaunch crazyflie_controller crazyflie2.launch frame:="body"
```

then the controller node starts.

whoenig, 2016-08-23:
The executable name is actually crazyflie_controller as well, so try
<code>rosrun crazyflie_controller crazyflie_controller</code>
The recommended usage is using launch files though, since this simplifies setting parameters etc.

antonioadaldo, 2016-08-23:
How did I miss that!
Thank you so much.

closed 2016-08-23 by antonioadaldo

srnetsr, 2018-02-02:
Hi, I have a programming question about this issue.
Actually, I am a beginner, so it will be very thankful if you explain specifically.

I looked over **hover_vicon.launch** file, and found the line that file="$(find crazyflie_controller)/launch/crazyflie2.launch".
Also, when I looked over the **crazyflie2.launch** file in crazyflie_controller/launch folder, I found that the line <node name="controller" pkg="crazyflie_controller" type="crazyflie_controller" output="screen">.
However, I do not understand this line, because there is no crazyflie_controller file in crazyflie_controller/src folder, but controller.cpp file...
Even I cannot find crazyflie_controller file at all.

Could you explain about it?

Thank you in advance.

whoenig, 2018-02-05:
`crazyflie_controller` refers to the package name and not a file name. The package name is (just in this case) identical to the folder name.

srnetsr, 2018-02-06:
Thank you for your reply, @whoenig .
In fact, the text editor did not show "the line" what I attached.
The line in crazyflie2.launch file is the following
< node name="controller" pkg="crazyflie_controller" type="crazyflie_controller" output="screen" >

According to this line, I think the package name is defined in "pkg", so package name is what you said, **crazyflie_controller**.
But, it also said type has the same name with pkg name....
That is, type is defined as "crazyflie_controller".
This is what I do not understand... the file "crazyflie_controller" does not exist...but controller.cpp exists in crazyflie_controller package...
Could you please explain about it for me?

Thank you.

whoenig, 2018-02-06:
This refers to the name of the executable defined [here](https://github.com/whoenig/crazyflie_ros/blob/master/crazyflie_controller/CMakeLists.txt#L100-L102).

netkimjh, 2018-02-07:
Thank you for clear explanation.

But, I leant from tutorial that they use like the following
add_executable(controller controller.cpp).
Then, you can use 
< node name="controller" pkg="crazyflie_controller" type="controller" output="screen" >

I am curious if you have any special reason for changing the name.
If there is no reason, then just ignore my question...

Thank you anyway.

whoenig, 2018-02-08:
No reason.