whoenig/crazyflie_ros #40: Multiple crazyradios

braraki, 2016-08-29:
Do you have any tips for connecting to multiple crazyradios through the ROS architecture? I would really appreciate it!

whoenig, 2016-08-30:
Just use the appropriate uri. For example `radio://0/90/2M` uses the first radio while `radio://1/100/2M` uses the second. Make sure that different radios use different channels, since they will be transmitting in parallel.

miaoyanan, 2017-08-01:
Hi@whoenig
I know use the multiple crazyradio by  changing  the first number of the uri .you know we can only change the radio channel and the radio address by the crazyflie client .but where can I change the first number.
Thank you very much!

whoenig, 2017-08-01:
the number simply refers to the n'th radio plugged into your PC. There is no configuration needed.

neudinglei, 2017-08-03:
@whoenig 
Hello !
I have recently done more unmanned aerial vehicles in collaborative control experiments and have encountered some questions that you can help.
First describe a phenomenon: with a radio control three crazyfile 2.0 fixed-point flight, when the distance between UAVs is small (for example, less than 1m), the UAV will be in shock. But once the interval between unmanned aerial vehicles increases, the shock will be reduced a lot. (Single UAV fixed point control, the effect is very good). With a radio control two crazyfile 2.0 fixed-point flight, even if the interval is less than 0.3m, almost no shock. Can you explain the above?
In addition, I have a few questions about radio:
1.Are there frequent warnings of connection quality that affect the control of the aircraft?
2.a radio control several aircraft best?
3.How does a computer communicate with multiple aircraft through two radio?
I hope to hear from you soon, thank you！

whoenig, 2017-08-04:
No double-posting please, see issue #17.

closed 2018-01-26 by whoenig