whoenig/crazyflie_ros #42: Crazyradio timeout when transmitting to multiple crazyflies

phoenixinwater, 2016-09-21:
Hey Whoenig, thank you for the work you have put into this system.

Currently I am getting libusb_timeout errors when I attempt to run multiple crazyflies off the one crazyradio. Physically this results in the radio having a solid red led.

Additionally, when running crazyflies off different radios I have been getting some strange behaviour.

I was hoping you had an idea for either of these issues and could also provide what versions of firmware for the radios and crazyflies that you use.

Thank you very much

Ryan

whoenig, 2016-09-21:
Hi Ryan,
Which branch are you using?
Libusb_timeout: I assume you have just one crazyflie_server running (e.g. using the multi_teleop_xbox360.launch file)?
Strange behavior: Did you configure different addresses for each Crazyflie?

I use the default radio firmware (I believe that is 0.53 or 0.54) and tested with the 2016.02 firmware release. I believe newer firmwares work as well, but I didn't try it myself yet.

phoenixinwater, 2016-09-21:
Hi Whoenig,

I am using the master branch.

Yes, one crazyflie_server. My set-up is essentially your multi_hover_vrpn.launch demo.
On that topic, I am getting an error with the crazyflieSDK argument of the multi_hover_vrpn.launch file, how is this parameter used?

I did not set up different addresses for the Crazyflies acting on separate radios, I will give that a shot, thank you.

Great, thank you.

whoenig, 2016-09-22:
The crazyflieSDK argument is not used anymore - it should be deleted. The different addresses are important to avoid crosstalk between the Crazyflies.

phoenixinwater, 2016-10-05:
Hi Whoenig, 

Sorry for the delay between messages.

I have updated the firmware of my devices to what you recommended. I am still looking into this further, but when setting unique addresses for the crazyflie v2s I have the following:

- The 'scan' tool returns the firmware version of the radio, but does not detect crazyflies with addresses. Or, the address of a crazyflie will change in the 'configure' of cfclient but when read using 'tools scan' or with cfclient 'scan', the detections will not include the address.
- This error when trying to run 'hover' with a set address:
```
[ERROR] [WallTime: 1475652001.141857] bad callback: <bound method Controller._joyChanged of <__main__.Controller instance at 0xb6b84e8c>>
Traceback (most recent call last):
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/topics.py", line 711, in _invoke_callback
    cb(msg)
  File "/home/jan/catkin_ws/src/crazyflie_ros/crazyflie_demo/scripts/controller.py", line 48, in _joyChanged
    value = int(rospy.get_param("ring/headlightEnable"))
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/client.py", line 460, in get_param
    return _param_server[param_name] #MasterProxy does all the magic for us
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/msproxy.py", line 123, in __getitem__
    raise KeyError(key)
KeyError: 'ring/headlightEnable'
```
Any assistance or ideas would be great. Thank you very much.

closed 2016-10-15 by phoenixinwater