whoenig/crazyflie_ros #43: using ps3 controller with ros

nicefoxj1, 2016-09-24:
Hello im having a trouble at using ps3 controller on crazyflie ros 
i installed cflib and python client and checked it running  successfully 
and i installed crazyfile ros and tried rosrun crazyfile_tools scan and found the address of my crazyfile
but when i try 
kyujin@kyujin-Samsung-Desktop-System:~/catkin_ws$ roslaunch crazyflie_demo teleop_ps3.launch uri:=radio://0/80/250K 

then i get a error like this
ERROR: cannot launch node of type [crazyflie_demo/controller.py]: can't locate node [controller.py] in package [crazyflie_demo]
(viz and sensor reading are normal only controls are malfuntioning)
i checked a launch file  controller.launch and found default arg for crazyfileSDK was set to
<arg name="crazyflieSDK" default="/home/whoenig/projects/crazyflie/crazyflie-clients-python/lib"/>

but i found that i don't have such path crazyflie-clients-python/lib
list of contents in my  crazyflie-clients-python is 

(venv) kyujin@kyujin-Samsung-Desktop-System:~/crazyflie-clients-python$ ls -l
total 104
-rw-rw-r-- 1 kyujin kyujin   587  9월 25 02:33 appveyor.yml
drwxrwxr-x 2 kyujin kyujin  4096  9월 25 02:33 bin
-rw-rw-r-- 1 kyujin kyujin 15468  9월 25 02:33 bitcraze.ico
-rw-rw-r-- 1 kyujin kyujin   225  9월 25 02:33 CMakeLists.txt
-rw-rw-r-- 1 kyujin kyujin  2236  9월 25 02:33 CONTRIBUTING.md
-rw-rw-r-- 1 kyujin kyujin   725  9월 25 02:33 CREDITS.txt
drwxrwxr-x 2 kyujin kyujin  4096  9월 25 02:33 examples
-rw-rw-r-- 1 kyujin kyujin 18415  9월 25 02:33 LICENSE.txt
-rw-rw-r-- 1 kyujin kyujin   180  9월 25 02:33 MANIFEST.in
-rw-rw-r-- 1 kyujin kyujin    56  9월 25 02:33 module.json
-rw-rw-r-- 1 kyujin kyujin  1071  9월 25 02:33 package.xml
-rw-rw-r-- 1 kyujin kyujin  7529  9월 25 02:33 README.md
-rw-rw-r-- 1 kyujin kyujin  1901  9월 25 02:33 RELEASE_CHECKLIST.md
-rw-rw-r-- 1 kyujin kyujin  5484  9월 25 02:33 setup.py
drwxrwxr-x 7 kyujin kyujin  4096  9월 25 02:34 src
drwxrwxr-x 3 kyujin kyujin  4096  9월 25 02:33 tools
drwxrwxr-x 2 kyujin kyujin  4096  9월 25 02:33 win32install

please help me solve this prob..

thanks for your wonderful work and thanks in advance

whoenig, 2016-09-25:
The crazyflieSDK is not used anymore - everything is now implemented in C++. I just pushed a change to reflect that (this will not have any functional change though).
The controller.py node it is missing is in crazyflie_demo/scripts/controller.py (https://github.com/whoenig/crazyflie_ros/blob/master/crazyflie_demo/scripts/controller.py). Please verify that you have this file on your local disk. Also, make sure that the file has the executable permission set (this is done automatically if you use git, but not if you just download an archive from github.)

nicefoxj1, 2016-09-25:
thanks whoenig the problem was executable permission as you said

whoenig, 2016-09-25:
Great!

closed 2016-09-25 by whoenig