whoenig/crazyflie_ros #45: Information of when Crazyflie connection is ready

ataffanel, 2016-10-13:
For the UWB LPS I need to set parameters in the Crazyflie at connection time. It is for setting the anchor positions. However parameters cannot be set before the Crazyflie is actually connected so I would need to know when the connection is actually done (ie. when "Ready ..." is printed in the log).

I can implement that myself but I am not really sure how to do it in the most "ROS way".

whoenig, 2016-10-13:
This issue was addressed in another branch (part of the Crazyswarm project), but not yet in the master branch. The basic idea is the following:
1. In the crazyflie_driver advertise the update_parameter service only after it is "connected" (There is no connection state in crazyflie_ros, however it is possible to upload parameters after the TOC has been downloaded.)
2. In the client code first wait_for_service and then use the service normally. (see e.g. https://github.com/whoenig/crazyflie_ros/blob/master/crazyflie_demo/scripts/controller.py).

I just pushed a change to the master branch which implements that logic. Let me know if that works for you.

ataffanel, 2016-10-17:
Thanks a lot! This is a much nicer way of doing it than what I could imagine doing. I will test and let you know.

ataffanel, 2016-10-20:
I can confirm that this is working, I can now upload the anchor postion at connection time (https://github.com/bitcraze/lps-ros/blob/master/scripts/lps_ekf_bridge.py#L34-L47). Thanks!

closed 2016-10-20 by ataffanel

whoenig, 2016-10-20:
Excellent. Thanks for testing and providing feedback!