whoenig/crazyflie_ros #50: How to choose radio URIs for multiple copter flight

adaldo, 2016-12-04:
Hi!

I was wondering what the different "pieces" of the radio URIs should be set for multiple copter flight:
```
radio://<radio_num>/<channel>/<address>
```
For the sake of simplicity, let's say that we are flying two CFs with a single CR.
If I understand correctly, then I can leave the `<radio_num>` to zero for both CFs.
Then I have to pick a `<channel>` and an `<address>` for each CF.
But then:

- Assuming that the two different CF should have two different URIs, is it ok to just set different `<channel>`s, or should I (also) set different `<address>`es?

- How do I pick a `<channel>`? Are there any constraints?
(For example: "it has to be a positive integer less than 100".)
- And how do I pick an `<address>`? Is there a format constraint?

- Is there a place in the Bitcraze wiki or tutorials that I am missing where I can read this things up?

Thanks in advance!

whoenig, 2016-12-04:
Ideally you use the same channel, but different addresses in your scenario. The channel is a number between 0 and 125. A good channel depends on your environment (the 2.4 GHz band is shared) and can be found with a script developed in our lab: https://github.com/bitcraze/crazyflie-lib-python/blob/master/examples/radio-test.py

You are correct about 0 for radio_num.

The address is a 40-bit number, in hex notation. There are no constraints on the address. The default is E7E7E7E7E7.

There will be a tutorial chapter in Springer's ROS book (to be released Spring 2017) with more details and step-by-step guide. If you contact me directly over E-Mail, I can send you a draft (for your personal use only).

closed 2016-12-04 by whoenig

ataffanel, 2016-12-04:
About the addresses there is some that should be avoided. The radio (nRF24LU1) specification contains this note:

> Addresses where the level shifts only one time (that is, 000FFFFFFF) can often be detected in
> noise and can give a false detection, which may give a raised Packet-Error-Rate. Addresses
> as a continuation of the preamble (hi-low toggling) raises the Packet-Error-Rate.

So anything that is not only made of 0, F, 5 or A should be alright.

adaldo, 2016-12-04:
Thanks, that was very informative.