whoenig/crazyflie_ros #52: joystick configuration

byronxubin, 2017-02-21:
Hi everyone,

After I have configured the crazyflie_ros and optitrack motion capture system, I have come up with a new problem, which is how to configure joystick. I do know that I need to install the hector_quadrotor package. But the thing is that I really don't know how to install this package on ROS Kinetic on ubuntu 16.04. 

My solution to install this package is `git clone <url> -b kinetic-devel`, then I use `wstool` to install all the dependencies which is described in the hector_quadrotor.rosinstall file. After all these, I went to the workspace to `catkin_make`, but after I have done all these, it still does not work. Does anyone have installed this package on kinetic? All help would be very appreciated.

whoenig, 2017-02-21:
This is how it should work (and I have it running on ROS Kinetic). I assume your `catkin_make` works fine? What issue are you seeing, i.e. how do you determine that it does not work?

byronxubin, 2017-02-21:
The error is as follows.
[image]

whoenig, 2017-02-21:
It looks like ws_tool got a ton of dependencies you don't need and you get a build error in something unrelated. Rather than fixing that, I suggest that you just copy the folders "hector_quadrotor_teleop" and "hector_uav_msgs" in your catkin workspace (and remove the unneeded dependencies) and then run `catkin_make` again on the subset.

byronxubin, 2017-02-21:
[image]

So there are still errors here. I only keep the "hector_quadrotor_teleop" and "hector_uav_msgs" in the src folder.

whoenig, 2017-02-21:
You can either add the hector_quadrotor_interface folder as well (which seems to have no additional dependencies according to package.xml), or use the files from the indigo-devel branch (which compile just fine with Kinetic).

byronxubin, 2017-02-21:
If I add "hector_quadrotor_interface" into the folder, then the `catkin_make` would fail again, the error message is like below.
[image]

whoenig, 2017-02-21:
Try `sudo apt install ros-kinetic-hardware-interface`

byronxubin, 2017-02-21:
Even I tired `sudo apt install`, it still cannot work since it says that cannot locate the package `ros-kinetic-hardware-interface`. I've met this problem earlier today.

whoenig, 2017-02-21:
It is part of the ROS repositories. How did you install ROS?

byronxubin, 2017-02-21:
The ROS installation is based on the tutorial on the ros.org, did I miss something?

byronxubin, 2017-02-21:
If you use `apt-cache search ros-kinetic-hardware-interface`, you would find nothing. So I am wondering that hardware interface is not a released package. So we cannot use `sudo apt-get` to install this package?

byronxubin, 2017-02-21:
@whoenig 
Hi Wolfgang,

I've just modified the devel folder of the workspace, in that I deleted some folders which created before when I used `catkin_make` command. After that, `catkin_make` works. The workspace build with no errors. Thank you for you instruction. By the way, I am wondering that do we have better way other than deleting the unuseful folder in the devel folder manually?

whoenig, 2017-02-21:
Glad you got it working. There is `catkin_make clean`, but this also just deletes the `devel` and `build` folders.

byronxubin, 2017-02-21:
But if you use `catkin_make` to build the workspace with errors, there is no way for you to use `catkin_make clean` right? I just tested `catkin_make clean`, you could still find floders in `./devel/share/`.

whoenig, 2017-02-21:
Not sure. You could use `catkin clean` if you want to use the catkin tools package (https://catkin-tools.readthedocs.io/en/latest/). This deletes the folders. Or you just delete the folders manually and rebuilt.

byronxubin, 2017-02-21:
Great. Thanks!

byronxubin, 2017-02-21:
@whoenig 
Hi, 

New issues here. After I have installed the two packages "hector_quadrotor_teleop" and "hector_uav_msgs". When I want to launch `teleop_xbox360.launch`, there is another error that it says the joy node cannot be launched. Does that mean that I need to install the joy package in the worksapce?

whoenig, 2017-02-21:
This is part of ROS. `sudo apt install ros-kinetic-joy` should install it.

bpospeck, 2017-08-23:
Since this was still open, and my problem relates, I figured it'd be cleaner to comment here instead of opening a new topic. 

I've been following along this thread to get hector_quadrotor installed and all associated dependencies so I can try the demo with my Crazyflie. This includes the joy package. I'm running ROS lunar on a Xubuntu 17.04 VM. I try to install ros-lunar-joy, but it is unable to locate the package. I've tried a couple of older distros (such as ros-kinetic-joy) with the same response. 

I wasn't able to find anything useful on google or the ROS site. Is this an issue with ROS, my VM, or what?

Thanks for your help!

whoenig, 2017-09-01:
According to the ROS webpage, joy should be released for lunar. Nevertheless, lunar was not tested on my side so far. You shouldn't mix packages from lunar and kinetic. If the package is really not released yet, you'll need to download the source code and put it in your workspace manually.

closed 2018-01-26 by whoenig

miaoyanan, 2018-04-30:
@whoenig 
Hello,
I have a problem with the node joystick, it is describe as followed.
**_Couldn't open joystick /dev/input/js0. Will retry every second._**
I am looking forward to your reply. Thanks.

whoenig, 2018-04-30:
Please open a new issue. This sounds like you have no joystick attached to your computer. After plugging it in, make sure it is recognized using, for example, `dmesg`.