whoenig/crazyflie_ros #53: How to get IMU data

byronxubin, 2017-03-14:
Hi Wolfgang,

Thanks for your contribution for this package. I have achieved hover and multi-point track using your  ros package. While I want to go further, I think I need to understand about the onboard controller. However, I found really little information about the onboard controller. Here, I want to get information about the imu, I do find the rostopic IMU, but when I echo this topic, there is nothing printed. So how can I get these information? Moreover, how can I control the onboard controller. I really did not find much about that. 

Thanks a lot.
Bin

ataffanel, 2017-03-14:
Hi, I can maybe answer about the onboard controller. I have been using this controller bridge to make the Crazyflie internal PID controller controlled like the crazyflie_ros demo controller: https://github.com/bitcraze/lps-ros/blob/master/scripts/crazyflie_controller_bridge.py.

byronxubin, 2017-03-14:
@ataffanel 

Thanks for your response. After reading your code, I'm still confused about how to get the IMU data and get access to internal controller. I could see what Wolfgang did is to calculate the velocity of X,Y,Z,Yaw and then publish these message. While after that, I did not know how the internal controller would get these message. Thanks in advance.

ataffanel, 2017-03-14:
@byronxubin My answer was only about how to send setpoint to the internal PID controller. In the context of autonomous flight the velocity is calculated onboard by the state estimator (ie. the kalman filter) I am not aware of any way to send it from ROS to the Crazyflie.

byronxubin, 2017-03-14:
@ataffanel Thanks. That actually clarifies my confusion.

byronxubin, 2017-03-15:
@ataffanel Sorry for bothering you again. After reading the crazyflie_server.cpp again, I'm still confused at the consequence I met. I noticed that we have a rostopic IMU being published, while nothing is published in real case. I found the parameter for `enable_logging_imu` is true. How can I actually publish these data? Thanks a lot.

ataffanel, 2017-03-15:
@byronxubin The IMU topic is not required to send setpoint.

I just tested quickly and the imu topic seems to be working the same way it always did:
```
$ rostopic echo /crazyflie/imu
header: 
  seq: 36433
  stamp: 
    secs: 1489565981
    nsecs: 244978494
  frame_id: crazyflie/base_link
orientation: 
  x: 0.0
  y: 0.0
  z: 0.0
  w: 0.0
orientation_covariance: [-1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity: 
  x: 0.00720959527036
  y: 0.00054752621949
  z: -0.00110738165248
angular_velocity_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
linear_acceleration: 
  x: -0.335517583005
  y: 1.64005131215
  z: 9.56861816704
linear_acceleration_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
---
```

byronxubin, 2017-03-15:
@ataffanel Are you just using crazyflie_ros package? When I try to use the same way, there is nothing to be printed. I can see '/crazyflie/imu' topic by using 'rostopic list', but still I  can get nothing. Which part do you think might go wrong?

ataffanel, 2017-03-16:
I have used this launchfile for the test: https://github.com/bitcraze/lps-ros/blob/master/launch/dwm_loc.launch

Are you sure you are connected to a Crazyflie? Does the M4 LED blinks on Crazyflie and the GREEN LED blinks on Crazyradio?

byronxubin, 2017-03-16:
@ataffanel I'm pretty sure about connecting to a Crazyflie. I am using crazyflie_ros package to instruct Crazyflie to track different setpoints. But I cannot get the imu data by using 'rostopic echo'. I think the connection is correct since I can give command to Crazyflie to take off and land.

byronxubin, 2017-03-17:
@whoenig Hi, I have tested this again today, there is not data published from rostopic '/crazyfile/imu` and `/crazyflie/battery`, I guess other topic related to crazyfile also not publish data. But I do find code about published data in `crazyflie_server.cpp`.

closed 2017-03-17 by byronxubin