whoenig/crazyflie_ros #54: Problem while conntecting two CFs with one CrazyRadio Dongle

byronxubin, 2017-04-04:
Hi @whoenig ,

Recently I want to hover two CFs with one CrazyRadio Dongle, while there is no error and the progress seems fine, like I can see two CFs in the visualization window. But I can clearly see that the CFs are not really connected since the light on CrazyRadio Dongle is still red which is different from normal case. So it is obvious that there is no response to the takeoff command. Do you have any idea about why it would cause that? Do I need to flash new firmware, cause I see other issues then I flashed Crazyflie2-nrf-firmware to my CFs, still cannot connect.

By the way, it works fine separately. Thanks in advance.

ataffanel, 2017-04-04:
Are you sure you are using the correct URI when connecting the two Crazyflies?

byronxubin, 2017-04-04:
I guess the two CFs have the same URI, like `radio:=/0/80/2M/E7E7E7E7`. Is there any way to modify to set different address?

whoenig, 2017-04-04:
Each CF needs to have a unique address. A tutorial chapter will be published in the ROS Springer Book (Volume 2). You can find a pre-print on my labs webpage: http://usc-actlab.github.io/publications.html.

byronxubin, 2017-04-04:
Thanks. I did the same thing as you wrote in your book, but when I change the Radio Address in Crazyflie 2.0 config, for example `0xE7E7E7E701`, then click write, but the address would turn back in to `0xE7E7E7E7E7`. So I guess the CF's address is still the default one. But I can reset radio channel using the pc client.

byronxubin, 2017-04-05:
Tried several times. But I still cannot change the address of Crazyflie. It turns out that the pc clinet cannot actually write the modified address.

whoenig, 2017-04-05:
Try connecting your CF over USB to reconfigure it. After reconfiguration, make sure you reboot the CF.

byronxubin, 2017-04-05:
Great! After trying several times, including flash `crazyflie-firmware` and `crazyflie2-nrf-firmware`, finally the address can be modified. Then what confuses me is that, before updated the firmware, the behavior of Crazyflie looks good since I have changed some part of your original package. While after updating, the hovering quality and way point tracking quality seems bad, it looks like that the CF oscillates a lot than before. I'm not sure why is that. Maybe I need to do some tuning tomorrow.

neudinglei, 2017-08-02:
@whoenig @binxxx 
Hello !
I have recently done more unmanned aerial vehicles in collaborative control experiments and have encountered some questions that you can help.
First describe a phenomenon: with a radio control three crazyfile 2.0 fixed-point flight,  when the distance between UAVs is small (for example, less than 1m), the UAV will be in shock. But once the interval between unmanned aerial vehicles increases, the shock will be reduced a lot. (Single UAV fixed point control, the effect is very good). With a radio control two crazyfile 2.0 fixed-point flight, even if the interval is less than 0.3m, almost no shock. Can you explain the above?
In addition, I have a few questions about radio:
1. Are there frequent warnings of connection quality that affect the control of the aircraft?
2. a radio control several aircraft best?
3. How does a computer communicate with multiple aircraft through two radio?
I hope to hear from you soon, thank you！