whoenig/crazyflie_ros #55: Adding custom log block

fjctp, 2017-06-02:
I want to log additional information, like motor inputs, from Crazyflie. It seems that this package provides a way to add custom logBlock (see `crazyflie_driver/crazyflie_server.cpp at line 309`), but there is no details on that. Therefore, I wonder how can I use that?

Thank you

ataffanel, 2017-06-08:
You can look at this launch file for example: https://github.com/whoenig/crazyflie_ros/blob/master/crazyflie_demo/launch/customLogBlocks.launch

You will need to first discover what log variable to log, for example with the Crazyflie client, then you can setup log-blocks to be sent back at regular interval. The data will be available as a ROS topic.

fjctp, 2017-06-11:
Thank you, it is working great!

closed 2017-06-11 by fjctp