whoenig/crazyflie_ros #56: Altitude Hold

suarez-ramon, 2017-07-10:
Hello,

I am just starting with the CF quadrotor and I had some questions. Is it possible with this implementation to maintain a given altitude and control linear velocities in x and y independently?  Or do i have to send commands in thrust as well if I'm publishing on the cmd_vel topic?

whoenig, 2017-07-11:
The included controller is a PID position controller, controlling roll/pitch/yaw and thrust independently. You could just keep the thrust component and write a custom control portion for x and y. If you use the controller, you don't need to publish on the cmd_vel topic (there is a goal-topic instead).

suarez-ramon, 2017-07-11:
So I could just use the thrust component in the controller and send a goal in height then send command directly to cmd_vel topic to control pitch, roll and yaw separately?

whoenig, 2017-07-12:
You can't use cmd_vel and goal at the same time. You can, however, copy the code from the current controller, keep the thrust component and compute the other values to your liking.

closed 2018-06-05 by whoenig