whoenig/crazyflie_ros #57: High frequence outuput of the controlle

DarioMariniello, 2017-07-17:
Hi!
first, I want to thank you for this project, because it is the fundament of the experimental part of my master thesis! Be sure that I will cite you in my report :)

I hope you could help me with the following issue with your PID. I won't go deep in what my application actually is, but basically, I'm running the controller at 60Hz and I'm publishing way points at 20hz with distance variable from the Crazyflies, but always on the same plane(mostly, on a circular path). I'm using your standard setting for the PID parameters, I only increased the boundaries of the control output from [-10,10] to [-30,30] (degrees?). The problem is in the shape of the control output that I get, that is shown in the figure attached: the blue and the red are respectively the X and Y PID outputs. The Zone 1 is a time that goes from the take-off to the reach of an initial static hovering point. Zone 2 consists of the time in which the Crazyflie is hovering, then in Zone 3, I start publishing a moving goal.
Questions: do you have any clue of why do I get so much high-frequency chattering? Moreover, even if I was reaching saturations and I had to increase the max amplitudes (and in spite of this I still get some saturation), (i know that the saturation could normally mean that gains are too high or the goal is too fast), the crazyflie doesn't seem to behave "aggressively", because it is naturally filtering a command signal like this. Do you have any idea of what could be the cause of this? How "normal" is an output like this from your PID?

Thanks in advance!

[image]

whoenig, 2017-07-18:
The included controller is a very basic PID controller. Those oscillations could come either from real oscillations of the CF, or jitter created by your Mocap system (which can happen for those small UAVs if the cameras are far away). I don't think I ever plotted the control output (I usually look at desired vs. true state), but my best guess is that this is unfortunately "normal".

I would recommend using the Crazyswarm project instead, even if just flying a single CF. You would need a custom firmware, but a much better non-linear controller runs on-board. It also supports ellipsoid motions and waypoint navigation, see https://github.com/USC-ACTLab/crazyswarm for details.

Good luck with your thesis - I am glad you choose the CF platform for testing!

DarioMariniello, 2017-07-18:
Thank you! Both the answer and the suggestions are really useful. I'm actually flying three\four crazyflies with the same radio and also implementing in some cases, in a really dense formation (I also have implemented collision avoidance), so the project crazyswarm looks perfect. The only problems are that I use a Qualisys Mocap,  do you think i can easily adapt the code?Moreover, I run my code on ROS indigo..is it going to be a problem in your opinion? Thanks a lot!

whoenig, 2017-07-18:
Does Qualisys have a C++ Linux SDK which allows to get the 3D points of markers? If though it would be easy to integrate. I believe it compiles on ROS Indigo just fine - we switched to Kinetic during development in our lab so we didn't test the latest version with Indigo.

whoenig, 2018-01-26:
Closing this issue. If you need help with Qualisys integration for the crazyswarm, please open an issue there.

closed 2018-01-26 by whoenig