whoenig/crazyflie_ros #58: Crazyflie loosing stability and not hovering

The-SS, 2017-07-24:
Hi,
I have been trying to get the Crazyflie to hover for a while, but it is always becoming unstable and either rotating around itself in midair then crashing, or accelerating and crashing in the walls.
I am using VICON to track the Crazylie. I am modifying the arguments of the static tf to align the base-link frame with the fixed frame. I am setting the requested x,y, and z destination coordinates to (0, 0, 0.5), and launching hover_vicon.launch. The Crazyflie usually elevates, appears to be approaching a hovering state, but then looses stability and starts to spin around itself or accelerates upwards before crashing.
Any idea why this might be happening?

Thank you.

whoenig, 2017-07-24:
Since you seem to be starting out, I suggest using the Crazyswarm instead (even if you just want to fly a single drone): https://github.com/USC-ACTLab/crazyswarm. It has a much better controller running on board the CF that will get much better tracking performance, support for trajectories etc.

The-SS, 2017-07-25:
Ok. I will use Crazyswarm instead.
Thank you.

closed 2017-07-25 by The-SS