whoenig/crazyflie_ros #59: controlling using another C++ ROS node

alexge233, 2017-07-26:
Hi,

This isn't an issue, but rather a set of questions.
I've been going through the source code, and understand that most of what I need is in `crazyflie_cpp` and `crazyflie_driver`.

I want to either use the shared library you've built (in which case my ros node will link against it), or preferably run a separate ros node (c++) and communicate with the crazyflie via messaging.

I can't figure out which nodelet or service I need to bring up. I looked at the demos, but they're all in python and assume the use of a joystic/controller.

I imagine what I'm trying to achieve is as simple as subscribing to your topics and publishing to `cmd_vel`?
In which case what nodes do I need to bring up?

Alternatively, how can I directly control it via c++ when including and linking with the crazyflie_cpp library? by that I mean reading the IMU values, and setting the same `cmd_vel` options.

PS: many thanks for the work you've done!

Regards,
Alex

whoenig, 2017-07-26:
All you need is `crazyflie_server` (and short living nodes or service calls to add CFs at runtime). If you prefer to not use ROS and/or built into one binary, you can take a look at https://github.com/whoenig/crazyflie_ros/blob/master/crazyflie_driver/src/crazyflie_server.cpp to see an example on how to use `crazyflie_cpp` directly. The `crazyflie_tools` folder has smaller examples using `crazyflie_cpp` only as well.

alexge233, 2017-07-27:
Thanks @whoenig I'll try with a ros node first and see if I can control it with messages

dani199221, 2017-08-01:
Hi @whoenig. I have a similar-ish issue. What I wanna do is make the crazyflie hover at some height (lets say 1m from ground) without using any external sensors like camera or positioning systems. Is there any way I can use your pid controller (`controller.cpp`) to do  this? I have not seen your `controller.cpp` in great depth but it looks like it uses external sensors to detect its position (/goal topic of type `geometry_msgs/PoseStamped`) to adjust itself.

whoenig, 2017-08-01:
This is not possible with the on-board sensors (imagine you try to hover the CF using a joystick while closing your eyes.) It might be possible to add additional sensors, such as a camera or the new flow deck to improve autonomous behavior.

dani199221, 2017-08-03:
@whoenig  Thanks for the quick reply. Is there any way i can use my keyboard to make it hover using the loco positioning system? I have been trying to do this for a couple of days now but cant figure it out. Any help would be greatly appreciated.

whoenig, 2017-08-04:
I haven't tried it most recently. Most of the loco position system support is back in the CF client. There used to be some support for ROS, but using the "old" mode, where ROS would compute the position estimate. Newer firmwares can do that on-board (assuming you tell them where the anchors are).
This is certainly something we would like to support (and have documentation on), but it is simply not there yet. You might be able to do that directly with the CF client, otherwise pull-requests are very welcome:-)

ataffanel, 2017-08-04:
The Loco Positioning System is supported by ROS even with the internal position calculation (using and EKF), there is example launch files in the lps ros package, for example: https://github.com/bitcraze/lps-ros/blob/master/launch/dwm_loc_ekf_hover.launch, and there is some setup doc there: https://wiki.bitcraze.io/doc:lps:ros.

The system is setup the same way as when the position was computed by ROS and the controller was in ROS: there is one node getting position from the EKF and publishing it in TF and one node accepting position setpoint and transmitting it to the Crazyflie (in this implementation the x/y/z position setpoint is send in roll/pitch/thrust, not the cleanest but it works). 

@dani199221 Starting from the _dwm_loc_ekf_hover.launch_ file and replacing the _joystick_controller_ and _pose_ nodes you should be able to control the hover point using a keyboard.

closed 2017-09-15 by alexge233