whoenig/crazyflie_ros #60: Extending crazyflie_cpp and crazyflie_driver to handle generic CRTP packets for new application protocols.

JGSuw, 2017-07-31:
Hello,

First I would like to say that I appreciate the work you have done, I have found the crazyflie_driver very useful for quickly getting my crazyflies integrated with ROS. I am interested in extending upon the crazyflie_cpp package to support a generic packet handler. Let me explain my motivation:

Currently I am using the crazyflie_driver and indirectly crazyflie_cpp for closed loop control with a motion capture system. Soon I will be transitioning to an on-board system for sensing and estimating body frame velocity.  In order to do integration and testing I will require sending various commands to the crazyflie and transmitting telemetry from a crazyflie to a ground control system in an asynchronous manner (i.e. not all communication is initiated by the ground control computer). 

Sending commands seems easy enough with crazyflie_cpp; but receiving asynchronous telemetry is another matter. It is my thinking that this does not fit well into the log blocks / parameter model, but would be easier to achieve with a generic CRTP packet handler. 

A straightforward and minimal implementation would be to en-queue CRTP packets which are not matched (by port and channel) to any of the logging / parameter protocol responses. These en-queued data could be published to a topic advertised by the crazyflie_driver node. This would achieve the goal of intercepting and then disseminating generic packets to other nodes in the system.

Would anyone like to discuss this? I'm not entirely sure about where this functionality ought to be implemented.

whoenig, 2017-08-01:
Sounds like a good plan to me. You could add a new callback function which can be registered, similar to https://github.com/whoenig/crazyflie_ros/blob/master/crazyflie_cpp/include/crazyflie_cpp/Crazyflie.h#L123-L126. Then you call that function (if it was set) here https://github.com/whoenig/crazyflie_ros/blob/master/crazyflie_cpp/src/Crazyflie.cpp#L419, instead of printing an error. Finally, in https://github.com/whoenig/crazyflie_ros/blob/master/crazyflie_driver/src/crazyflie_server.cpp you register a new callback and publish some custom ROS message inside the callback.

I think this could be pretty useful for developers, similar to the custom log-message subsystem.

JGSuw, 2017-08-01:
Thanks for getting back to me so quickly!

I follow along with what you are saying, but I have a concern. From what I can tell, all of the incoming libusb transfers happen in the Crazyradio::sendPacket method, following an outgoing transfer. It is my understanding that using this approach only a single packet will be received for each outgoing packet. 

If this is the case, then if the crazyflie is sending packets asynchronously, not all of the incoming packets will be transferred through libusb, because they did not follow an outgoing packet from the ground control computer. I would expect this to cause some buffer to overflow, and packets to be lost.

If this is indeed a problem, then a simple way to solve it is to periodically transfer empty packets to the crazyflies, thereby triggering an incoming bulk transfer. However, this seems like a waste of resources.

Another solution would be to replace the synchronous libusb transfers with [asynchronous submissions](http://libusb.org/static/api-1.0/group__asyncio.html). This would involve more modifications to the code. For one, heap memory would have to be allocated for the transfers, since the memory would have to be valid outside the scope of the sendPacket method. Also, the main executable loop need to somehow initiate [polling of the libusb file descriptors](http://libusb.org/static/api-1.0/group__poll.html), so that the libusb events can be handled.

Do you think that this will be an issue, and if so do you know of a work around?

whoenig, 2017-08-01:
You made excellent points. However, what you describe is not supported on the hardware side. The radio (or any nrf chip) can operate in two modes: Primary Receiver (PRX) and Primary Transmitter (PTX). Currently, the firmware is always in PRX-mode and the radio is always in PTX mode. That means, the CF can actually only "send" data when it is piggybacked as payload to an ACK. That's why there is this constant ping to gather data. Additionally, thats why the CF firmware never really "sends" data, but puts everything in queues (which are then emptied whenever an incoming message needs an ack.)
In principle it would be possible to switch PTX/PRX roles (even dynamically), but this would require significant changes on the firmware sides of CF and the radio.

JGSuw, 2017-08-02:
Thanks for that information, I did not know about the firmware limitations of the radio.

I am making good progress thus far on getting packets published. Once I have it tested some more, I will fork your repositories (crazyflie_cpp, and crazyflie_driver) and open a pull request for you to check out. I am thinking there will be two feature branches on each; one for publishing packets from the crazyflie, and a second branch which allows the user to transmit packets to the crazyflie via a service call to the crazyflie_server.

closed 2018-01-26 by whoenig