whoenig/crazyflie_ros #63: How does controller.cpp receive the postion information and how to generate the control input

zhanghaijason, 2017-08-29:
Hi whoenig,
I read the code again recently and still have several questions.
For the hovering problem:
1. How could ROS know CF's position? I know it should be related to targetDrone,

Code: Select all

tf::StampedTransform transform;
                m_listener.lookupTransform(m_worldFrame, m_frame, ros::Time(0), transform);

                geometry_msgs::PoseStamped targetWorld;
                targetWorld.header.stamp = transform.stamp_;
                targetWorld.header.frame_id = m_worldFrame;
                targetWorld.pose = m_goal.pose;

                geometry_msgs::PoseStamped targetDrone;
                m_listener.transformPose(m_frame, targetWorld, targetDrone);

                tfScalar roll, pitch, yaw;
                tf::Matrix3x3(                                       //tf namespace, Matrix3x3, class
                    tf::Quaternion(                                  //tf namespace, Quaternion, class
                        targetDrone.pose.orientation.x,
                        targetDrone.pose.orientation.y,
                        targetDrone.pose.orientation.z,
                        targetDrone.pose.orientation.w
                    )).getRPY(roll, pitch, yaw);

Which sentence actually gets the CF's position? How could it get the position data? In my understanding, the vicon/optitrack node should publish the position data to a topic A and the controller.cpp subscribes to topic A to get the position data. But I did not find such publisher in vicon's code, either the subscriber in controller.cpp
2. For the PID controller
The input to these controllers are the desired hover position and the current feedback. Is it realistic to get the desired control input(pitch, roll, yaw rate, thrust) from the error one to one?(I mean control input roll is generated only from x position error, desired control input pitch is generated only from y position error... ).

Thanks

whoenig, 2017-08-29:
1. vicon_bridge tracks the position relative to /world and publishes transforms. See https://github.com/whoenig/crazyflie_ros/blob/master/crazyflie_demo/launch/hover_vicon.launch#L4 for the frame used.
The lookupTransform function gets the pose (position and orientation).
2. This is a very simple controller. There are much better non-linear controllers, one of which is implemented as part of the Crazyswarm project in a custom CF firmware.

zhanghaijason, 2017-09-04:
Hi @whoenig, I use the mocap_optitrack package to get the pose feedback. After `rosrun tf view_frames` , the pdf I got is 
[frames.pdf](https://github.com/whoenig/crazyflie_ros/files/1275247/frames.pdf). 
And `roslaunch crazyflie_demo hover_vicon.launch
uri:=radio://0/100/2M/E7E7E7E701 frame:=crazyflie1 x:=0 y:=0
z:=0.5
rosrun tf view_frames`, what I got is 
[whole_frames.pdf](https://github.com/whoenig/crazyflie_ros/files/1275249/whole_frames.pdf)

I can see the position of CF in RVIZ, and the coordinate seems correct. However, I cannot make it take off with ps3. I tried the teleoperation example and there is no error. 

Are the frame figure correct? Do you have any idea how to solve this? 
Thanks

whoenig, 2017-09-05:
Your tf frames look correct. There might be an issue with your PS3 joystick. Try looking at the topic: `rostopic echo /crazyflie/joy` and see if the output changes if you move the ps3 sticks. Also, for PS3 joysticks you frequently need to press the PS3 button in the middle after you connected them, otherwise they wont work. Similarly, check the output of  `rostopic echo /crazyflie/imu` to see if it is just a visualization issue in rviz, or if there is really no data coming from the CF.

zhanghaijason, 2017-09-06:
@whoenig  The PS3 can work, there is always output when I move the sticks. However,  for the hovering demo, there is no output for /crazyflie/imu and for the teleop_ps3, there is output, but all the data is 0 when the imu arrow doesn't show up in RVIZ.
In addition, the stuck of CF in RVIZ is because the bad marker positions in which condition the optitrack cannot track CF.

zhanghaijason, 2017-09-15:
Hi @whoenig , I bought a new CF and now it can take off. However, I failed again for the hovering experiment. The CF always flies into wrong direction and finally it goes out of the optitrack's vision scope. Do you have any idea to solve this
[image]
. The optitrack I use is [http://optitrack.com/products/v120-trio/](url). The vision scope is not wide. Like the attach file

zhanghaijason, 2017-09-22:
Hi @whoenig 
I have several questions
1. The ROS sometimes gives me the info `Size doesn't match!`, why is it?
2. For the controller. If the target in  CF's body frame is (3,-1,0). Then according to the CF's left hand frame, the roll<0, pitch<0 should be correct. 
On the other hand, for the y direction in CF's body frame in controller.cpp, linear.y=PID(0,-1), and the error in PID controller is defined as error=-1-0=-1, however, the PID parameter of Y direction are <0, then the linear.y>0, roll=linear.y>0, which is opposite to what it should be(roll<0).

whoenig, 2017-09-22:
1. What data do you try to log? I believe this can happen if you try to log too many items per log request (i.e. if the data doesn't fit into the 32 byte acknowledgement packet).
2. In the firmware roll (y) is somewhere inverted for the setpoint; I forgot the exact place, but that is why there is a different sign for pitch vs roll.

zhanghaijason, 2017-09-22:
@whoenig 
Thanks for your explanation. The first question is because I tried to log x,y,z position in CF's body frame and also the control input. 
In addition, previously I failed many times for the hovering experiment. It is because the markers I use are two heavy for the CF, so it influences the CF's stabilization.  Considering my motion tracking system is not that powerful, the CF may fly out of optitrack's vision field.  Hope this can help those who also place too heavy markers on CF.

zhanghaijason, 2017-09-27:
Hi @whoenig , I am trying to implementing a nonlinear controller stated in Crazyswarm for hovering and for my project in the future. Do you have any suggestions how to tune the PID parameters?

prbhtm, 2018-03-29:
@zhanghaijason Please let me know the changes you made in hover_optitrack.launch

s3xyxavy, 2018-04-02:
@prbhtm hi, I am also using optitrack as my external positioning system. Would like to help if you have any issues.

prbhtm, 2018-04-03:
Thanks, @s3xyxavy. I modofied hover_vicon.launch file correctly. I can give take off and land commands from my joy stick. My crazyflie does not stay at a fixed point but it moves in some direction. Is this because of coordinate mismatch with optitrack and ros?

s3xyxavy, 2018-04-03:
1. Put your 4 markers on 3 legs and 1 on top 

2. Turn 90 degree left/right and re- Rigid body it on Optitrack 

3. Re launch the hover file

s3xyxavy, 2018-04-04:
There is a coordinate and orientations mismatch. May I know what streaming node are you using? https://github.com/ros-drivers/mocap_optitrack Should work fine.

prbhtm, 2018-04-04:
@s3xyxavy , I am using vrpn client. Unfortunately mocap and natnet don't give any output when I type rostopic echo /crazyflie/pose; rosrun tf view_frames also gives no data received. Initially both were working. Now I am able to connect only by vrpn client. Therefore, I am thinking to focus on vrpn and correct the coordinate and orientation mismatch. DO you have any idea how to do it?

s3xyxavy, 2018-04-04:
[image]
Access this part of the file: https://github.com/ros-drivers/vrpn_client_ros/blob/kinetic-devel/src/vrpn_client_ros.cpp 
And have it changed to this Axis conversion. [Take note of the negative sign]
Always check your IP address on your VRPN, as it changes once in awhile. Yes and I recommend you use the VRPN client, its more 'editable'.

prbhtm, 2018-04-04:
@s3xyxavy Thanks a lot. It is still diverging but slowly. I am also getting an error related to link quality
[ WARN] [1522860195.066634571]: Link Quality low (0.170000)

I am using ROS-Indigo on ubuntu 14.04 and my crazy radio PA is 0/80/250K . I get link quality warning even if I put crazyflie in front of crazy radio.

s3xyxavy, 2018-04-04:
Yea I get that link quality too, but as Long it loads the takeoff emergency and landing you are good to Go. 

You should try facing the red led to the positive x of the optitrack and re-rigid body on Optitrack again.

s3xyxavy, 2018-04-10:
@prbhtm Hi, try facing your crazyflie 2.0 red led to the positive of Optitrack's X-axis and Re-Rigid body the marker on Optitrack. My convention only works in Y-up on Optitrack.

prbhtm, 2018-05-12:
@s3xyxavy Do you know the closed-loop dynamics of crazyflie? I understand that it is linear but I need to calculate time constants for each axis and yaw.

podnerogato, 2018-11-28:
I have a similar question as I want to publish these roll/pitch/yaw control inputs into a node so i can bag them. How to do that?
I have accessed and published cmdVelChanged which are said to be the roll/pitch/yaw references in the crazyflie_server.cpp, but these are actually position references(?). Is it possible to publish them from controller.cpp and how? If I am not mistaken, these values are published in m_pubNav which doesn't appear on a rostopic list when using dwm_loc_ekf_hover.launch. I have tried writing a new node but the written publisher doesn't appear on the rostopic list neither.

podnerogato, 2018-11-29:
Thank you for the fast replay. Can you point me to the files i can modify by writing a publisher for the data i need?

zhanghaijason, 2019-09-02:
> @zhanghaijason Please let me know the changes you made in hover_optitrack.launch

I am really sorry to reply you so late. I don't have a notification. Sorry. But I am happy to see you solved this.

gpc668, 2020-12-05:
hello everyone ! I have some questions when i use the crazyflie_demo ,when i roslaunch crazyflie_demo hover_vrpn.launch, it can linked,but when i use the take off button ,the motors runs but the rotate speed don't enough for the crazyflie to fly.
How can i solve the problem?