whoenig/crazyflie_ros #66: Support for Velocity Set-Points with Crazyflie Flow Deck

JGSuw, 2017-09-16:
I was wondering, since bitcraze has now released their [optical flow expansion dec](https://www.bitcraze.io/flow-deck/)k for Crazyflie 2.0, it would be nice to do higher level control (with body-frame velocity setpoints, as opposed to attitude setpoints) in ROS, and to let the Crazyflie internally handle the velocity request to attitude loop internally. For example, the cmd_vel topic which is used by the crazyflie_server could be reinterpreted to be a velocity request, whereas currently it is an attitude request (and thrust).

Does this functionality currently exist in crazyflie_ros? If not, I think it would be an excellent addition. For many people, it would be a more natural starting point for autonomy and off-board control of the Crazyflie.

whoenig, 2017-09-16:
This is not yet supported, but I have a student working on adding support and perhaps a python scripting layer to allow scripted behaviors without mocap.

JGSuw, 2017-11-17:
Hello,

I just wanted to revisit this idea, now that I have some more time to hack on my crazyflie. Looking at crazyflie-lib-python, it appears that there two velocity set-point types which can be transmitted to the commander module via the CRTP protocol.

I'm speaking in particular about 'velocity_world_setpoint'(s) and 'hover_setpoint'(s). Both of these packets contain x and y velocity set-points, but not z velocity. Z velocity may not be controllable without some z distance feedback, but bodyframe velocity requests may be implementable by just sending the corresponding packet to the commander module.

I will try some basic scripted behavior through crazyflie_driver using the SendPacket service.

whoenig, 2018-04-18:
This is now available with the `cmd_hover` and `cmd_position` topics.

closed 2018-04-18 by whoenig

YunzeLi725, 2018-05-25:
Hi Whoenig,

I am wondering if crazyflie_ros package support the Flow Deck now? If so, which demon can be launched with the Flow Deck equipped?
Thanks.

whoenig, 2018-05-25:
Yes, the flow deck is fully supported. There is nothing special to take care of from the ROS side - the flow deck will simply improve the onboard state estimation. You can, for example, execute waypoint following or trajectories without a motion capture this way.

whoenig, 2018-05-30:
Yes, that's possible. There is an example script here: https://github.com/whoenig/crazyflie_ros/blob/master/crazyflie_demo/scripts/Hover.py.
Another option is to use the high-level mode, which allows you to follow waypoints or execute trajectories.