whoenig/crazyflie_ros #69: High frequency streaming

d-i-a-s, 2017-10-06:
Hello, 

I have been working on high frequency streaming from the Crazyflie and using the python library to communicate with it. I'm now starting to work with this packages since there was previous work done with them in my university.
I have changed the firmware of the Crazyflie so it starts to send packets as soon as is turned on.

When I connect it to the server, after the deck messages are displayed, I start getting the message "Don't know ack: Port: 1 Channel: 0 Len: 31" that I think is issued because I don't have a method handing this kind of packets. After this error is displayed around 80 times I get this error.

`***** Internal Program Error - assertion (i < N) failed in T& boost::array<T, N>::operator[](boost::array<T, N>::size_type) [with T = unsigned char; long unsigned int N = 30ul; boost::array<T, N>::reference = unsigned char&; boost::array<T, N>::size_type = long unsigned int]:
/usr/include/boost/array.hpp(123): out of range
[crazyflie_server-1] process has died [pid 13917, exit code -6, cmd /home/david/projects/catkin_cf/devel/lib/crazyflie_driver/crazyflie_server __name:=crazyflie_server __log:=/home/david/.ros/log/1f51758e-aa6b-11e7-8921-50b7c387376c/crazyflie_server-1.log].
log file: /home/david/.ros/log/1f51758e-aa6b-11e7-8921-50b7c387376c/crazyflie_server-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
`

I know this error is somehow connected to the publishing of packets (when I commented this [line](https://github.com/whoenig/crazyflie_ros/blob/master/crazyflie_driver/src/crazyflie_server.cpp#L166) is didn't crashed, which doesn't make much sense to me since there in lines 162 and 163 you are accessing the same structure. At first I though it was because the vector was not thread safe or the saving of the packets was eating too much memory, but now I noticed you are copying it so I'm not sure of what is causing the problem.

Thank you

d-i-a-s, 2017-10-06:
I think found the error but I'm still not sure why it was crashing in the first place. 

In this [line](https://github.com/whoenig/crazyflie_ros/blob/master/crazyflie_driver/src/crazyflie_server.cpp#L164) you do the loop until i < packet.size but it should be done until i < (packet.size-1) since you have removed the header already. Right?

But I'm sending 30 bytes of data + 1 byte (header) so it shouldn't overflow (go above 31 in the array index). When I do echo on the packets topic I have this output after I add the -1:
```
size: 31
header: 16
data: [248, 233, 126, 158, 233, 158, 150, 233, 238, 163, 233, 126, 154, 233, 142, 159, 233, 254, 154, 233, 158, 158, 233, 14, 157, 233, 94, 155, 233, 176]
```
Everything looks fine, I'm just not sure why it was crashing in the first place. In the Crazyflie firmware I'm setting the size to 30 bytes and not 31, I didn't thought I should count with the header, can this be the problem?

d-i-a-s, 2017-10-06:
Just found out why it was crashing... In the crtpPacket.msg the data is defined to be 30 bytes long. It probably should be 31 but since no one can use it...
Sorry for all the comments, I guess the only thing that needs to be fixed is the addition of the -1.

whoenig, 2017-10-06:
Thanks for the report! Where do you think we'll need a "-1"? I believe packet.size is the number of data bytes (i.e. without the header byte).

d-i-a-s, 2017-10-09:
In the Crazyflie firmware I'm setting the size to 30 (I'm not counting with the header byte) but as seen above, when the packets are published the size is 31. My guess is that the size is incremented somewhere in the firmware but I wasn't able to find where. Following this logic I would say that the size takes into account the header byte. 

I needed and the -1 in the for condition in this [line](https://github.com/whoenig/crazyflie_ros/blob/master/crazyflie_driver/src/crazyflie_server.cpp#L164).
```
      std::vector<Crazyradio::Ack>::iterator it;
      for (it = packets.begin(); it != packets.end(); it++)
      {
        crazyflie_driver::crtpPacket packet;
        packet.size = it->size;
        packet.header = it->data[0];
        for(int i = 0; i < (packet.size-1); i++)
        {
          packet.data[i] = it->data[i+1];
        }
        m_pubPackets.publish(packet);
      }
```

closed 2017-10-18 by d-i-a-s