whoenig/crazyflie_ros #70: Run time error for flying multiple crazyflies

anchekuri, 2017-10-12:
Hi I am using crazyflie_ros to fly 2 crazyfllie with one CR dongle . I have the same radio channel set for both the crazyflies and have a deferent address.  

I am using the command to launch as 
roslaunch bitcraze_lps_estimator dwm_loc_ekf_multi_hover.launch uri0:=radio://0/30/2M/E7E7E7E701 uri1:=radio://0/30/2M/E7E7E7E702

But I am having the issue when I run the command that the 2 crazyflies are not connecting and showing error status in the rviz.
And I am missing 

[INFO] [1507778222.452673]: Anchor 1 at [0, 2, 1.85]
terminate called after throwing an instance of 'std::runtime_error'
  what():  Could not create log block! Result: 204 1 205 110
Traceback (most recent call last):

And different  error about the log file every time I run the command too..

I hope my terminal screenshot can explain you more

[image]

Can I know hoe to rectify the log error (run time error). 
And fly the 2  crazyflies .

Thank you ,

Appala Chekuri

whoenig, 2017-10-12:
It complains that "acc.x" is not available on your firmware.

1. What firmware are you using?
2. Does it work when you work with a single CF (each of them individually)?

anchekuri, 2017-10-12:
@whoenig 

Thank you, For your reply.

I tried to fly both the CF individually with these commands 
roslaunch bitcraze_lps_estimator dwm_loc_ekf_hover.launch uri:=radio://0/30/2M/E7E7E7E701

roslaunch bitcraze_lps_estimator dwm_loc_ekf_hover.launch uri:=radio://0/30/2M/E7E7E7E702

**and they are flying individually( it was working).**

Here I am using dwm_loc_ekf_hover.launch file instead of dwm_loc_ekf_multi_hover.launch.

About my firmware of the crazyflie . I updated the latest firmware for lps_node, crazyflie, nrf-firmware and the radio-firmware. 
I think the issue is with the dwm_loc_ekf_multi_hover.launch file.
If you want I can show you the code of the dwm_loc_ekf_multi_hover.launch that I didnt changed the file except the radio channels and JS1 instead of JS0 for joystick controller.
 And then I moved the anchor_pos.yaml .example to anchor_pos.yaml file .

I didnt change nothing more than that in the lps-ros code  .

and any help will be appreciated. 
Thank you,

Appala Chekuri.

whoenig, 2017-10-12:
I looked at the dwm_loc_ekf_multi_hover.launch file and didn't see any issue with it. In particular, the file for a single CF configures the CFs exactly the same way as the multi-version does. You can try disabling some of the additional logging variables, in particular set `enable_logging_imu` to False within the crazyflie_add call (a full list of options and their default values is here: https://github.com/whoenig/crazyflie_ros/blob/master/crazyflie_driver/launch/crazyflie_add.launch#L10-L15).

anchekuri, 2017-10-13:
@whoenig 
Thanks for your previous reply. I have tried the way what you mentioned in the previous reply but still has the same issue .

I also installed ubuntu 16.04.03 again and updated the latest crazyflie-firmware with kalman filters added to the tools/make/config.mk and then compiled and upload to both CF's.

Then followed the instructions on the link https://wiki.bitcraze.io/doc:lps:ros for localization as it is 

and then :-  

I tried both keeping the logging variables to "False" for all the logging variables.

I was showing the same error with mentioning that 
**ERROR:-
first time :-** Could not create the logging file
then I tried to run it again and I encountered with another error as
**second time** :-  terminate called after throwing an instance of 'std::runtime_error'
  what():  Could not find gyro.z in log toc!

But later I changed the logging variables to "True" and then added the radio addresses of the crazyflies 
to this and I added the code screenshot here 

[image]

Then I ran the command 

roslaunch bitcraze_lps_estimator dwm_loc_ekf_multi_hover.launch uri0:=radio://0/30/2M/E7E7E7E701 uri1:=radio://0/30/2M/E7E7E7E702

And I was encountered with the error (added screenshot)

[image]

But when I am trying to Fly the crazyflies individually they are working 

I am not sure with why I am encountering this error . 

Any help would be appreciated

Thank you ,

Appala Chekuri

whoenig, 2017-10-13:
1. How do you obtain your firmware and flash it? Do you use a released version or a self-compiled version?
2. You shouldn't edit crazyflie_add.launch. Instead, add your variables to your launch files, e.g. in dwm_loc_ekf_multi_hover.launch:

```
<node pkg="crazyflie_driver" type="crazyflie_add" name="crazyflie_add" output="screen">
      <param name="uri" value="$(arg uri0)" />
      <param name="tf_prefix" value="crazyflie0" />
      <param name="enable_logging" value="False" />
      <rosparam>
        genericLogTopics: ["log_kfpos", "log_kfqt"]
        genericLogTopicFrequencies: [30, 30]
        genericLogTopic_log_kfpos_Variables: ["kalman.stateX", "kalman.stateY", "kalman.stateZ"]
        genericLogTopic_log_kfqt_Variables: ["kalman.q0", "kalman.q1", "kalman.q2", "kalman.q3"]
      </rosparam>
</node>
```
3. Which version of crazyflie_ros are you using? The latest master-branch?

anchekuri, 2017-10-17:
@whoenig Sorry for the late reply. 

Thank you for your time and help.
1. https://github.com/bitcraze/crazyflie-firmware.git This was the  firmware I was using to flash the crazyflie.
2. I have made changes with the dwm_loc_ekf_multi_hover.launch and it was working without error but not accurately sometimes . like behaving crazy and not taking off or moving out of the field and hitting walls. 
3. For crazyflie Ros I was using the latest master-branch version.

I have some doubts about anchor setup.

Right now I have 6 anchors from 0-5. As mentioned in the https://wiki.bitcraze.io/doc:lps:tdoa  that I can use 8 anchors from 0-7. i added the 7th and 8th anchor to the anchor_pos.yaml. And when i run the command 
roslaunch bitcraze_lps_estimator dwm_loc_ekf_multi_hover.launch uri0:=radio://0/30/250K/E7E7E7E701 uri1:=radio://0/30/250K/E7E7E7E702
then is was showing that there was error with anchor 6  and anchor 7 which are 7 and 8.

And also I have a question about dwm_loc_ekf_swarm_hover.launch. I has a code for 5 CF's can I use less than 5 or more than 5 for that or any restrictions are there to this launch file. 

Thank you,
Appala Chekuri.

whoenig, 2017-10-18:
Hi Appala,
I don't have the Bitcraze UWB system setup at the moment. Perhaps you can ask those questions in the official support forum or open an issue at https://github.com/bitcraze/lps-ros for help?

whoenig, 2018-01-26:
Closing this issue here, please open a new issue if you think there is an issue in crazyflie_ros.

closed 2018-01-26 by whoenig