whoenig/crazyflie_ros #71: Changing LED Extension Ring Parameters

karenacai, 2017-10-14:
I purchased a CD LED-ring expansion which has the capability of changing the LED pattern depending on the /ring/effect parameter. 

In ROS, if I enter the command:  `roparam list`, it comes up as the parameter /crazyflie_23/ring/effect. If I check the parameter value of this parameter, it is originally 6, which corresponds to the LED having two white circles moving in half circles. If I change this parameter using the command `rosparam set /crazyflie_23/ring/effect 0`, I would expect the LED pattern to turn off. Instead, the same LED pattern (which corresponds to the parameter value 6) remains.

Is there a way to ensure that the CrazyFlie hardware is actually set when this ROS parameter is changed?

karenacai, 2017-10-14:
If there is no good solution for doing this, can you point to the location in the crazyflie_ros sourcecode where the ROS environment is getting these parameter values from the crazyflie hardware, and setting them in the ROS environment? That would be super helpful as well!

whoenig, 2017-10-14:
The parameter server does not notify clients if a value is updated. Thus, you will need to tell crazyflie_ros that you want a certain parameter (or a list of parameters) updated, by calling the `update_params` service. You can find an example here: https://github.com/whoenig/crazyflie_ros/blob/master/crazyflie_demo/scripts/controller.py#L48-L53

Note: dynamic_reconfigure might allow what you are looking for more elegantly, but at least a few years ago it didn't support a dynamic parameter list as required by the Crazyflie.

closed 2018-01-26 by whoenig