whoenig/crazyflie_ros #72: No Crazyflie over USB found while running (Crazyflie_tools scan)

cheikhnasy, 2017-10-14:
Hello ,
I am trying to use my crazyflie with ROS , when i run the command "rosrun crazyflie_tools scan
" i detect the radio but not the crazyflie.all my rules files are well set-up. What could be the problem?

Note: i am able to control my crazyflie using the virtual machine provided by bitcraze .

regards.

whoenig, 2017-10-14:
1. Do you have a rules file for the Crazyflie itself (i.e. `SUBSYSTEM=="usb", ATTRS{idVendor}=="0483",
ATTRS{idProduct}=="5740", MODE="0664", GROUP="plugdev"`)?
2. Are you using the latest master branch? The USB support was just recently pushed (it was part of a different branch before).

cheikhnasy, 2017-10-16:
Hello Whoenig,
 yes i have a rules file for the Crazyflie itself.And yes i am using the latest master branch

[image]. 

Thank you for your help!

cheikhnasy, 2017-10-17:
Hello, 
i just noticed that i detect  my crazyflie when i run the "crazyflie_tools scan" command with my crazyflie physically connected to my computer via a usb cable. can it be a matter of permission?  

Thank you!

whoenig, 2017-10-17:
Did you change the address of your CF? The scan tool can only find CFs with a known address (run `crazyflie_tools scan --help` for details on how to set the address. If you don't know the address, connect over USB and run the original CF client. Connect to the CF and select Configure 2.0 in the menu.

cheikhnasy, 2017-10-17:
Hello whoenig , 
I did not change any address. it is still the same one 0XE7E7E7E7E7 .

whoenig, 2017-10-17:
And you can find the CF and connect to it using the official python client?

whoenig, 2017-10-18:
This is very strange indeed. Could you try any of the other tools that require a known uri? For example, `rosrun crazyflie_tools listParams --uri radio://0/80/250K`.

I don't think it can be a permission issue (otherwise it wouldn't be able to find the radio either).

cheikhnasy, 2017-10-23:
Hello sorry for the late answer ,

I have tried other tools  and they didn't work . Then i connected my crazyflie to my iphone via bluetooth and now i can't any more find or connect it to the python client .is there any way to reset it? 

regards

ataffanel, 2017-10-24:
You can connect the Crazyflie to the python client using the USB port, this way you will be able to reconfigure the radio channel, datarate and address.

cheikhnasy, 2017-10-24:
Thank you so much ataffanel ! it is working ,yet i can still connect it using ROS, it looks dumb but can you tell me how i should exactly install the packages on my computer ? i use Indigo

Regards

whoenig, 2017-10-24:
There is a tutorial in the following book chapter: W. Hönig and N. Ayanian. "Flying Multiple UAVs Using ROS", Chapter in Robot Operating System (ROS): The Complete Reference (Volume 2), Springer, 2017. You can find a pre-print at http://act.usc.edu/publications.html

cheikhnasy, 2017-10-27:
Hello whoenig , 

sorry again for the late answer ! I followed the tutorial and tried other tools who worked , i just have a problem with the tool "scan".I am totally lost actually, do you have some suggestions for me ?

regards .
[image]

whoenig, 2017-10-27:
I am not sure what the problem is? It found your crazyflie (radio://0/100/250K) correctly. The other message just notes that you don't have a Crazyflie directly attached using a USB cable.

cheikhnasy, 2017-10-27:
Thank you for the quick answer , i just attached the crazyflie and as you can see ,i can detect it!
[image]

The-SS, 2017-12-20:
Hi.
I have a similar problem. 
When I run `rosrun crazyflie_tools scan` I get the following:
```
Found Crazyradio with version 99.55
No Crazyflie over USB found.
```
Even after flashing the cf firmware following [this](https://wiki.bitcraze.io/doc:crazyflie:dev:starting) and upgrading the crazyradio firmware following [this](http://crazyswarm.readthedocs.io/en/latest/usage.html), I still get the same issue.
I checked my `99-crazyflie.rules` file and it has the following content:
```
#Crazyflie
SUBSYSTEM=="usb", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="5740", MODE="0664", GROUP="plugdev"
```
When I run `rosrun crazyflie_tools listParams --uri radio://0/80/2M/E7E7E7E701`, I get the following output:
```
cpu.flash (uint16, readonly)
cpu.id0 (uint32, readonly)
cpu.id1 (uint32, readonly)
cpu.id2 (uint32, readonly)
system.selftestPassed (int8)
flightmode.althold (uint8)
flightmode.poshold (uint8)
flightmode.posSet (uint8)
flightmode.yawMode (uint8)
flightmode.yawRst (uint8)
flightmode.stabModeRoll (uint8)
flightmode.stabModePitch (uint8)
flightmode.stabModeYaw (uint8)
cmdrCPPM.rateRoll (float)
cmdrCPPM.ratePitch (float)
cmdrCPPM.rateYaw (float)
cmdrCPPM.angRoll (float)
cmdrCPPM.angPitch (float)
locSrv.enRangeStreamFP32 (uint8)
pid_rate.roll_kp (float)
pid_rate.roll_ki (float)
pid_rate.roll_kd (float)
pid_rate.pitch_kp (float)
pid_rate.pitch_ki (float)
pid_rate.pitch_kd (float)
pid_rate.yaw_kp (float)
pid_rate.yaw_ki (float)
pid_rate.yaw_kd (float)
pid_attitude.roll_kp (float)
pid_attitude.roll_ki (float)
pid_attitude.roll_kd (float)
pid_attitude.pitch_kp (float)
pid_attitude.pitch_ki (float)
pid_attitude.pitch_kd (float)
pid_attitude.yaw_kp (float)
pid_attitude.yaw_ki (float)
pid_attitude.yaw_kd (float)
sensorfusion6.kp (float)
sensorfusion6.ki (float)
sensorfusion6.baseZacc (float)
posEst.estAlphaAsl (float)
posEst.estAlphaZr (float)
posEst.velFactor (float)
posEst.velZAlpha (float)
posEst.vAccDeadband (float)
velCtlPid.vxKp (float)
velCtlPid.vxKi (float)
velCtlPid.vxKd (float)
velCtlPid.vyKp (float)
velCtlPid.vyKi (float)
velCtlPid.vyKd (float)
velCtlPid.vzKp (float)
velCtlPid.vzKi (float)
velCtlPid.vzKd (float)
posCtlPid.xKp (float)
posCtlPid.xKi (float)
posCtlPid.xKd (float)
posCtlPid.yKp (float)
posCtlPid.yKi (float)
posCtlPid.yKd (float)
posCtlPid.zKp (float)
posCtlPid.zKi (float)
posCtlPid.zKd (float)
posCtlPid.thrustBase (uint16)
posCtlPid.thrustMin (uint16)
posCtlPid.rpLimit (float)
posCtlPid.xyVelMax (float)
posCtlPid.zVelMax (float)
controller.tiltComp (uint8)
motorPowerSet.enable (uint8)
motorPowerSet.m1 (uint16)
motorPowerSet.m2 (uint16)
motorPowerSet.m3 (uint16)
motorPowerSet.m4 (uint16)
firmware.revision0 (uint32, readonly)
firmware.revision1 (uint16, readonly)
firmware.modified (uint8, readonly)
imu_tests.MPU6500 (uint8, readonly)
imu_tests.HMC5883L (uint8, readonly)
imu_tests.MS5611 (uint8, readonly)
imu_sensors.HMC5883L (uint8, readonly)
imu_sensors.MS5611 (uint8, readonly)
sound.effect (uint8)
sound.neffect (uint32, readonly)
sound.freq (uint16)
sound.ratio (uint8)
system.taskDump (uint8)
ring.effect (uint8)
ring.neffect (uint32, readonly)
ring.solidRed (uint8)
ring.solidGreen (uint8)
ring.solidBlue (uint8)
ring.headlightEnable (uint8)
ring.glowstep (float)
ring.emptyCharge (float)
ring.fullCharge (float)
anchorpos.anchor0x (float)
anchorpos.anchor0y (float)
anchorpos.anchor0z (float)
anchorpos.anchor1x (float)
anchorpos.anchor1y (float)
anchorpos.anchor1z (float)
anchorpos.anchor2x (float)
anchorpos.anchor2y (float)
anchorpos.anchor2z (float)
anchorpos.anchor3x (float)
anchorpos.anchor3y (float)
anchorpos.anchor3z (float)
anchorpos.anchor4x (float)
anchorpos.anchor4y (float)
anchorpos.anchor4z (float)
anchorpos.anchor5x (float)
anchorpos.anchor5y (float)
anchorpos.anchor5z (float)
anchorpos.enable (uint8)
```
So, `crazyflie_tools listParams` works, but `crazyflie_tools scan` does not not return the connected crazyflie.
What could be the issue?

whoenig, 2017-12-20:
The `scan` tool can only find crazyflies with known address (it scans for datarate and channel). Use `rosrun crazyflie_tools scan --address E7E7E7E701` instead. If you don't know the address of your CF, connect it using a uUSB cable and check the configuration using the Bitcraze Python Client.

The-SS, 2017-12-20:
I ran `rosrun crazyflie_tools scan --address E7E7E7E701` and got the following output 
```
Found Crazyradio with version 99.55
radio://0/80/2M/E7E7E701
No Crazyflie over USB found.
```
So, the crazyflie was found, yet `No Crazyflie over USB found` is still being printed out?

whoenig, 2017-12-20:
This just means that you didn't connect a CF using a physical USB cable. If you would attach your CF to your computer using a micro USB cable, this message would go away and it would tell you the uri (`usb://0` I believe) instead.

The-SS, 2017-12-20:
I have the CF connected via a USB cable and yet I'm getting that output.

whoenig, 2017-12-20:
You will need a different rules file for the operation over USB. You can find instructions in W. Hönig and N. Ayanian. "Flying Multiple UAVs Using ROS", Chapter in Robot Operating System (ROS): The Complete Reference (Volume 2), Springer, 2017, pages 5 and 6 in the preprint available at http://act.usc.edu/publications.html.

The-SS, 2017-12-25:
I've done everything as explained in your reference. I have two rules files: `99-crazyflie.rules` and `99-crazyradio.rules`.
`99-crazyradio.rules` contains the following:
```
#Crazyradio
SUBSYSTEM=="usb", ATTRS{idVendor}=="1915", ATTRS{idProduct}=="7777", MODE="0664", GROUP="plugdev"
#Bootloader 
SUBSYSTEM=="usb", ATTRS{idVendor}=="1915", ATTRS{idProduct}=="0101", MODE="0664", GROUP="plugde
```
`99-crazyflie.rules` contains the following:
```
#Crazyflie
SUBSYSTEM=="usb", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="5740", MODE="0664", GROUP="plugdev"
```
Both files have the the same content as those specified in your reference (W. Hönig and N. Ayanian. "Flying Multiple UAVs Using ROS", Chapter in Robot Operating System (ROS): The Complete Reference (Volume 2), Springer, 2017, pages 5 and 6). 
I reloaded the udev rules, logged out and back in, and ran the same command again with the crazyradio connected, yet I still got the message `No Crazyflie over USB found.`.

What should I do about that? 
And does this affect the operation of `crazyswarm` or `crazyflie_ros`  since the cf won't be connected via USB?

whoenig, 2017-12-26:
You could try rebooting, just in case. Also, check using `lsusb` if your Crazyflie shows up and has the specified vendor and product id.

This shouldn't affect any of the other functionality.

whoenig, 2018-01-26:
I pushed a fix that improves the output of the scan tool. Verbose output can now be enabled using the `-v` flag and is disabled by default.

closed 2018-01-26 by whoenig