whoenig/crazyflie_ros #73: Higher packege loss

d-i-a-s, 2017-10-18:
Hello,

As I said before in this [issue](https://github.com/whoenig/crazyflie_ros/issues/69) I'm working in high frequency streaming from the Crazyflie and up until a few weeks ago I was working with the bitcraze python library.

I'm streaming at around 370 packets/sec from the Crazyflie to the computer and as expected at higher rates I have some packets losses. My problem is that when using the python library the packet loss is at around 1-4% but when using crazyflie_ros is at 22-30%.

My questions are:
1. Do you have any idea what is causing this?
2. Can you explain why you choose to implement your own radio software to connect a Crazyflie to ROS rather than using the python library?

Thank you!

whoenig, 2017-10-18:
Hi,
That is a very interesting problem! I have no idea what might cause this, but let me elaborate on your second question first.

The python library is more feature complete, but doesn't support multiple Crazyflies well, for two reasons: a) there is no support for connecting to multiple CFs using a single radio, and b) there are serious performance issues when trying to control more than 5 CFs using the python lib. The C++ implementation fixes both issues. We have used it to control 49 CFs with just 3 radios in the Crazyswarm project (see http://crazyswarm.readthedocs.io/en/latest/). Also, I want to point out that the re-implementation is fairly straight-forward. In particular, the python lib uses bindings to lib-usb and crazyflie_ros uses lib-usb natively. Most of the low-level code is a 1-to-1 code transfer. Thus, I am not sure where a packet loss would come from.

How do you measure packet loss and which (low-level) functions in crazyflie_ros are you calling to send/receive packets? Did you look at RSSI (in both python and crazyflie_ros)?

Best,
Wolfgang

d-i-a-s, 2017-10-19:
I wasn't aware of much of that information. Thank you.

I measure packet loss by incrementing the value for the first byte of each packet. I keep track of that value for the previous received packet and by comparing the two I know if any packet were lost. This means that I have to receive the packets in order.

The packet data flow is:
1. Queued by queueGenericPacket function.
2. Published by publishPackets function.
3. I have [this](https://github.com/d-i-a-s/crazyflie_ros/blob/master/mic_deck/scripts/micDeck.py) python script which subscribes to the topic and then calls the [incoming](https://github.com/d-i-a-s/crazyflie_ros/blob/master/mic_deck/scripts/streamPort.py#L50) function for each packet published.

The code I use for the python library is pretty much the same except for the way it receives the data.

An important note which I found very odd: I had to add a condition to check if the incremented value was the same for two consecutive packets because the incoming function was being called twice for some packets.

d-i-a-s, 2017-10-19:
I'm so stupid... I'm sorry for wasting your time. I forgot to set the logging for the rest of the stuff (battery, temperature, ...) to false. Once I did this the packet loss percentage is pretty much the same as when using the python library.

Btw for some reason there is nothing being published in the /rssi topic. I haven't looked at it in either of them.

ataffanel, 2017-10-23:
Just a note @whoenig for your point a): There is now support for connecting multiple Crazyflie using a single or multiple radio in the python lib. There might still be performance problem with a massive amount of connection, but the functionality exists.

whoenig, 2017-10-23:
That's good to know - thanks Arnaud! I'll close this issue since it seems to be resolved now.

closed 2017-10-23 by whoenig