whoenig/crazyflie_ros #76: Access IMU data in controller.cpp

zhanghaijason, 2017-10-30:
Hello whoenig,
I am using you code to do CF  control. Now I want to estimate CF's speed using Kalman FIlter. And I want to get the IMU (x acceleration) in the code. Could you help me figure  out how to do this? I think it should be related to void onImuData(uint32_t time_in_ms, logImu* data) in cf_server.cpp. However, I am not good in C++, could you give me hint?

whoenig, 2017-10-30:
I don't think that the IMU data will help you that much: You can only get the data at 100 Hz and, depending on what other messages the CF sends back, there can be a significant delay between the measurements and the time you receive the data. You will achieve much better results if you implement state estimation and control on-board. Both crazyswarm (http://crazyswarm.readthedocs.io/en/latest/) and official crazyflie firmwares come with Kalman filters.

zhanghaijason, 2017-11-01:
@whoenig  Thanks. It turns out the update equation of my Kalman filter is wrong. Now I can estimate the speed. Even though it is not accurate enough, but is much better that using dx/dt

closed 2018-01-26 by whoenig