whoenig/crazyflie_ros #77: Crazyflie2 will be disconnected after a few minutes

virtualclone, 2017-11-02:
I use crazyflie_ros package to control Crazyflie2 (by Crazyradio PA)
and I use keyboard to control it (publish velocity to topic /crazyflie/cmd_vel
and after a few minutes , Crazyflie2 will be disconnected,why?

the error shows
```
ncrl@ncrl-VirtualBox:~/catkin_ws$ roslaunch crazyflie_demo teleop_xbox360.launch uri:=radio://0/80/2M
... logging to /home/ncrl/.ros/log/747a5772-bf76-11e7-ae67-0800273f37ab/roslaunch-ncrl-VirtualBox-10532.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.1.125:39101/

SUMMARY
========

PARAMETERS
 * /crazyflie/crazyflie_add/enable_logging: True
 * /crazyflie/crazyflie_add/enable_logging_battery: True
 * /crazyflie/crazyflie_add/enable_logging_imu: True
 * /crazyflie/crazyflie_add/enable_logging_magnetic_field: True
 * /crazyflie/crazyflie_add/enable_logging_packets: True
 * /crazyflie/crazyflie_add/enable_logging_pressure: True
 * /crazyflie/crazyflie_add/enable_logging_temperature: True
 * /crazyflie/crazyflie_add/pitch_trim: 0
 * /crazyflie/crazyflie_add/roll_trim: 0
 * /crazyflie/crazyflie_add/tf_prefix: crazyflie
 * /crazyflie/crazyflie_add/uri: radio://0/80/2M
 * /crazyflie/crazyflie_add/use_ros_time: True
 * /crazyflie/joy/dev: /dev/input/js0
 * /crazyflie/quadrotor_teleop/x_axis: 5
 * /crazyflie/quadrotor_teleop/x_velocity_max: 30
 * /crazyflie/quadrotor_teleop/y_axis: 4
 * /crazyflie/quadrotor_teleop/y_velocity_max: -30
 * /crazyflie/quadrotor_teleop/yaw_axis: 1
 * /crazyflie/quadrotor_teleop/yaw_velocity_max: -200
 * /crazyflie/quadrotor_teleop/z_axis: 2
 * /crazyflie/quadrotor_teleop/z_velocity_max: 60000
 * /rosdistro: kinetic
 * /rosversion: 1.12.7

NODES
  /crazyflie/
    crazyflie_add (crazyflie_driver/crazyflie_add)
    crazyflie_demo_controller (crazyflie_demo/controller.py)
    joy (joy/joy_node)
    quadrotor_teleop (crazyflie_demo/quadrotor_teleop)
  /
    crazyflie_server (crazyflie_driver/crazyflie_server)
    rqt_plot2 (rqt_plot/rqt_plot)
    rqt_plot3 (rqt_plot/rqt_plot)

ROS_MASTER_URI=http://192.168.1.125:11311

core service [/rosout] found
process[crazyflie_server-1]: started with pid [10550]
process[crazyflie/crazyflie_add-2]: started with pid [10552]
process[crazyflie/joy-3]: started with pid [10577]
process[crazyflie/quadrotor_teleop-4]: started with pid [10591]
process[crazyflie/crazyflie_demo_controller-5]: started with pid [10599]
process[rqt_plot2-6]: started with pid [10613]
process[rqt_plot3-7]: started with pid [10620]
[ INFO] [1509591897.091652470]: Opened joystick: /dev/input/js0. deadzone_: 0.050000.
[ INFO] [1509591897.092102626]: wait_for_service /add_crazyflie
[ INFO] [1509591897.093054276]: found /add_crazyflie
[ INFO] [1509591897.115247224]: Adding radio://0/80/2M as crazyflie with trim(0.000000, 0.000000). Logging: 1, Parameters: 1, Use ROS time: 1
Received unrequested data for block: 0
Received unrequested data for block: 0
Received unrequested data for block: 0
[... 133 lines cut ...]
Received unrequested data for block: 0
Received unrequested data for block: 0
Received unrequested data for block: 0
Received unrequested data for block: 0
Received unrequested data for block: 0
Received unrequested data for block: 0
Received unrequested data for block: 0
Received unrequested data for block: 0
Received unrequested data for block: 0
Received unrequested data for block: 0
Received unrequested data for block: 0
Received unrequested data for block: 0
Received unrequested data for block: 0
Received unrequested data for block: 0
Received unrequested data for block: 0
Received unrequested data for block: 0
terminate called after throwing an instance of 'std::runtime_error'
  what():  timeout
[crazyflie_server-1] process has died [pid 10550, exit code -6, cmd /home/ncrl/catkin_ws/devel/lib/crazyflie_driver/crazyflie_server __name:=crazyflie_server __log:=/home/ncrl/.ros/log/747a5772-bf76-11e7-ae67-0800273f37ab/crazyflie_server-1.log].
log file: /home/ncrl/.ros/log/747a5772-bf76-11e7-ae67-0800273f37ab/crazyflie_server-1*.log
[rqt_plot3-7] process has finished cleanly
log file: /home/ncrl/.ros/log/747a5772-bf76-11e7-ae67-0800273f37ab/rqt_plot3-7*.log
[rqt_plot2-6] process has finished cleanly
log file: /home/ncrl/.ros/log/747a5772-bf76-11e7-ae67-0800273f37ab/rqt_plot2-6*.log
```

whoenig, 2017-11-02:
How long does it take until it disconnects? It looks to me like it has trouble configuring the logging variables. You can try disabling logging. Also, make sure you reboot the CF before connecting - the "Received unrequested data" information probably stems from the fact that there was some old logging enabled.

virtualclone, 2017-11-07:
@whoenig thanks,I did what you said,and solved the problem.
but there are another problem.
I want to use Optitrack to hover the crazyflie2.0 at the point
I roslaunch hover_vrpn.launch,than use `rosservice call /crazyflie/takeoff`(because I don't have a joystick)
it will take off to the point I want , but after about 20 seconds,it will be disconnected.
the error show like this
```

ncrl@ncrl-VirtualBox:~$ roslaunch crazyflie_demo hover_vrpn.launch uri:=radio://0/100/2M/E7E7E7E701 x:=0 y:=0 z:=0.5
... logging to /home/ncrl/.ros/log/dccc40da-c398-11e7-bd31-0800273f37ab/roslaunch-ncrl-VirtualBox-3856.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.1.125:39613/

SUMMARY
========

PARAMETERS
 * /crazyflie/controller/PIDs/X/integratorMax: 0.1
 * /crazyflie/controller/PIDs/X/integratorMin: -0.1
 * /crazyflie/controller/PIDs/X/kd: 20.0
 * /crazyflie/controller/PIDs/X/ki: 2.0
 * /crazyflie/controller/PIDs/X/kp: 40.0
 * /crazyflie/controller/PIDs/X/maxOutput: 10.0
 * /crazyflie/controller/PIDs/X/minOutput: -10.0
 * /crazyflie/controller/PIDs/Y/integratorMax: 0.1
 * /crazyflie/controller/PIDs/Y/integratorMin: -0.1
 * /crazyflie/controller/PIDs/Y/kd: -20.0
 * /crazyflie/controller/PIDs/Y/ki: -2.0
 * /crazyflie/controller/PIDs/Y/kp: -40.0
 * /crazyflie/controller/PIDs/Y/maxOutput: 10.0
 * /crazyflie/controller/PIDs/Y/minOutput: -10.0
 * /crazyflie/controller/PIDs/Yaw/integratorMax: 0.0
 * /crazyflie/controller/PIDs/Yaw/integratorMin: 0.0
 * /crazyflie/controller/PIDs/Yaw/kd: -20.0
 * /crazyflie/controller/PIDs/Yaw/ki: 0.0
 * /crazyflie/controller/PIDs/Yaw/kp: -200.0
 * /crazyflie/controller/PIDs/Yaw/maxOutput: 200.0
 * /crazyflie/controller/PIDs/Yaw/minOutput: -200.0
 * /crazyflie/controller/PIDs/Z/integratorMax: 1000.0
 * /crazyflie/controller/PIDs/Z/integratorMin: -1000.0
 * /crazyflie/controller/PIDs/Z/kd: 6000.0
 * /crazyflie/controller/PIDs/Z/ki: 3500.0
 * /crazyflie/controller/PIDs/Z/kp: 5000.0
 * /crazyflie/controller/PIDs/Z/maxOutput: 60000.0
 * /crazyflie/controller/PIDs/Z/minOutput: 10000.0
 * /crazyflie/controller/frame: /crazyflie
 * /crazyflie/controller/worldFrame: world
 * /crazyflie/crazyflie_add/enable_logging: False
 * /crazyflie/crazyflie_add/enable_logging_battery: True
 * /crazyflie/crazyflie_add/enable_logging_imu: True
 * /crazyflie/crazyflie_add/enable_logging_magnetic_field: True
 * /crazyflie/crazyflie_add/enable_logging_packets: True
 * /crazyflie/crazyflie_add/enable_logging_pressure: True
 * /crazyflie/crazyflie_add/enable_logging_temperature: True
 * /crazyflie/crazyflie_add/pitch_trim: 0
 * /crazyflie/crazyflie_add/roll_trim: 0
 * /crazyflie/crazyflie_add/tf_prefix: crazyflie
 * /crazyflie/crazyflie_add/uri: radio://0/100/2M/...
 * /crazyflie/crazyflie_add/use_ros_time: True
 * /crazyflie/pose/name: goal
 * /crazyflie/pose/rate: 30
 * /crazyflie/pose/x: 0
 * /crazyflie/pose/y: 0
 * /crazyflie/pose/z: 0.5
 * /rosdistro: kinetic
 * /rosversion: 1.12.7

NODES
  /crazyflie/
    baselink_broadcaster (tf/static_transform_publisher)
    controller (crazyflie_controller/crazyflie_controller)
    crazyflie_add (crazyflie_driver/crazyflie_add)
    pose (crazyflie_demo/publish_pose.py)
  /
    crazyflie_server (crazyflie_driver/crazyflie_server)

ROS_MASTER_URI=http://192.168.1.125:11311

core service [/rosout] found
process[crazyflie_server-1]: started with pid [3874]
process[crazyflie/crazyflie_add-2]: started with pid [3876]
process[crazyflie/controller-3]: started with pid [3888]
process[crazyflie/pose-4]: started with pid [3907]
process[crazyflie/baselink_broadcaster-5]: started with pid [3912]
[ INFO] [1510044727.940532045]: wait_for_service /add_crazyflie
[ INFO] [1510044727.953389341]: found /add_crazyflie
[ INFO] [1510044727.981292917]: Adding radio://0/100/2M/E7E7E7E701 as crazyflie with trim(0.000000, 0.000000). Logging: 1, Parameters: 0, Use ROS time: 1
[ INFO] [1510044728.205962764]: Requesting parameters...
Params: 99
[crazyflie/crazyflie_add-2] process has finished cleanly
log file: /home/ncrl/.ros/log/dccc40da-c398-11e7-bd31-0800273f37ab/crazyflie-crazyflie_add-2*.log
[ INFO] [1510044733.481980626]: Ready...
[ INFO] [1510044733.482103129]: Elapsed: 5.367817 s
[ INFO] [1510044735.182050792]: Takeoff requested!
terminate called after throwing an instance of 'std::runtime_error'
  what():  LIBUSB_ERROR_TIMEOUT
[crazyflie_server-1] process has died [pid 3874, exit code -6, cmd /home/ncrl/catkin_ws/devel/lib/crazyflie_driver/crazyflie_server __name:=crazyflie_server __log:=/home/ncrl/.ros/log/dccc40da-c398-11e7-bd31-0800273f37ab/crazyflie_server-1.log].
log file: /home/ncrl/.ros/log/dccc40da-c398-11e7-bd31-0800273f37ab/crazyflie_server-1*.log
```

whoenig, 2017-11-07:
I am not sure what the cause of this might be. Perhaps it is a side-effect of using a virtual machine. As workaround, you could comment out the lines `throw std::runtime_error(libusb_error_name(status));` in https://github.com/whoenig/crazyflie_ros/blob/master/crazyflie_cpp/src/Crazyradio.cpp (3 occurrences). There are other checks in higher-levels which should gracefully deal with errors from a single failed libUSB call. If that doesn't help, I recommend trying a native installation.

virtualclone, 2017-11-10:
@whoenig ,  I use the native installation , and it works , thank you very much!!

closed 2017-11-11 by whoenig