whoenig/crazyflie_ros #78: hover with vrpn problems

virtualclone, 2017-11-10:
1.
I use vrpn to hover crazyflie2.0 at a given point
I roslauch the` hover_vrpn.launch`
and it takes off successfully to the point (x,y,z)=(0,0,0.5)
but it starts to drift in the z axis after about 10 seconds,then it will fly out of control
2.
Can I just publish the position to the goal topic to change the position after takeoff?
```
rostopic pub /crazyflie/goal geometry_msgs/PoseStamped "header:
  seq: 0
  stamp:
    secs: 0
    nsecs: 0
  frame_id: ''
pose:
  position:
    x: 0.5
    y: 0.0
    z: 0.5
  orientation:
    x: 0.0
    y: 0.0
    z: 0.0
    w: 0.0" -r 10
```
I try to use publish like this after it hover at (x,y,z)=(0,0,0.5)
but it seems doesn't work,it start to fly out of control(perhaps it is from the problem1)

whoenig, 2017-11-11:
1. You should check the battery voltage to make sure. If the battery is low, there might be erratic behavior. Does the tracking work correctly or do you lose tracking?
2. Yes, changing the topic will move the quadrotor. You should avoid large step-inputs though. There is an example for waypoint following included.

virtualclone, 2017-11-16:
the tracking works correctly,
and when the battery voltage is lower than half of the voltage,it starts to fly unstable.
Is it normal?
thanks for your reply

whoenig, 2017-11-16:
I think it depends on the payload and "structure" you added to fix some markers on the frame. Also, the controller in crazyflie_ros is not very good (it is not a sophisticated controller and runs externally). A better one is the on-board controller included in the crazyswarm project (see http://crazyswarm.readthedocs.io/en/latest/). This one can fly without issues until the batteries are nearly empty (and then the CF just falls down).

virtualclone, 2017-11-17:
thanks,I get it.I'll try crazyswarm.

closed 2018-01-26 by whoenig